panoptes/POCS #26: Supervisord

wtgee, 2016-05-05:
Review supervisord scripts and make sure we are starting all items appropriately.

- [ ] Weather scripts (#25)
- [ ] Sensors scripts (#25)
- [ ] PAWS site
- [ ] Unit script
- [ ] Other? (e.g. maintenance scripts, data processing, updating, etc.)

wtgee, 2017-02-03:
We need to review this and see if it is what we want. The idea was to have these items start up automatically after reboot. However it seems like reboots should only happen manually in which case the user wants to probably start things up manually (maybe with one simple command). If a reboot happens unexpectedly we should probably alert the user somehow and allow them to inspect the system before start things.

jamessynge, 2017-09-09:
I'd like a full startup script, with flags to control what is attached, and then the user can decide whether to run that script automatically at boot. For manual control, perhaps we could have it start a screen or tmux session, with separate "window" for each of the shells that needs to be executed (e.g. peas_shell).

wtgee, 2018-04-03:
Closing this as stale.

closed 2018-04-03 by wtgee