panoptes/POCS #29: Camera Simulator

wtgee, 2016-05-06:
Camera simulator needs to be able to produce images, including pointing images (ideally with some offset). Need a folder of test images to work with this.

joshwalawender, 2016-05-06:
One way to do this without needing a folder of images, would be to have a tool which would generate an "image" by dropping gaussian stars down based on catalog positions.  Something like this might already exist, so it would be worth looking around for one.

joshwalawender, 2016-05-12:
Here's something that looks like it might do the trick.  It's from the same developer as SExtractor.
http://www.astromatic.net/software/skymaker

wtgee, 2016-05-12:
Nice, thanks. We can either use this or one or two real images. We just
need to be able to plate solve them then move them a tiny bit to check
tracking/pointing etc.

Hopefully I'll be getting to this soon.

wtgee, 2017-02-03:
There are currently some dummy images in the test folder (`$POCS/tests/data`) and the camera simulator will just always use these as a file name. This allows for unning of the entire machine with a camera simulator. Additional features should be defined separately.

closed 2017-02-03 by wtgee