panoptes/POCS #39: Accelerometer Positioning

wtgee, 2016-06-29:
We want to be able to recover the pointing position of the mount based on the accelerometer readings located inside the camera box. Various ways to do this exist. See e.g. (in no particular order):

http://perso-etis.ensea.fr/~pierandr/cours/M1_SIC/AN3397.pdf
http://arxiv.org/abs/1407.0035
http://www.it.bton.ac.uk/staff/gw4/papers/Displacement%20from%20Accelerometer.pdf

Values are available directly off the accelerometer (stored in mongodb locally) or off BigQuery. Values take the form:

```json
                "camera_box" : {
                        "accelerometer" : {
                                "o" : 7,
                                "x" : -0.96,
                                "z" : -0.16,
                                "y" : -0.15
                        }
                }
```
where `o` represents the orientation (which can determine which side of the pier the accelerometer currently is)

astroblogweb, 2017-06-19:
Is this issue still open? Can I work on the python code for it?

joshwalawender, 2017-06-19:
Hi @astroblogweb , we have a summer intern who is just starting today and will be working on this topic.  Perhaps we can find another area yoiu might be interested in, or perhaps you can work together on it.  @wtgee coordinates all our software and he is away at a conference this week, so we might be a bit slow replying here for the next few days.

wtgee, 2023-05-23:
Currently not recommending the accelerometer in the build so closing this as stale.

closed 2023-05-23 by wtgee