panoptes/POCS #42: Daily Housekeeping

wtgee, 2016-06-29:
Determine tasks that need to be accomplished during the Housekeeping state, which is run after the unit is parked. This should either be run directly after parking or at some set time each morning.

jamessynge, 2018-04-24:
I'm looking at setting up some code to run PIAA analysis on an observation, which would be easier if the last file uploaded for an observation was a sentinel that indicates that the upload is complete. This could be a metadata file, or just an empty file (e.g. UPLOAD-COMPLETE).

jamessynge, 2018-04-25:
Closing the issue right after I added a feature request, without a comment
as to why, leaves me wondering whether the feature request is already
recorded elsewhere, is already handled, or you disagree. Could you
elaborate?

wtgee, 2018-04-25:
Sorry, I didn't mean to close it. In fact, I had made an earlier comment and had meant to post it but guess I screwed it up on my mobile.

Anyway, yes, I have been thinking something along these lines. Because they can be uploaded in any order I almost think it would need to be a metadata file that lists the names of the files that should have been uploaded. If we wanted to be really sure we could do md5sums or something.

I have just barely got it set up so that it responds to the notifications when something gets uploaded. The idea is that this will then do a plate-solve on the file and ingest metadata from the FITS header into a postgresql database. We can then use this metadb for displaying a site of info.

jamessynge, 2018-04-25:
Thanks for the clarification. My goal is somewhat simpler: trigger the
light curve production once all files of an observation have been uploaded.
As you say, a metadata file can include the names, and maybe the
signatures, of the raw files, so a responder can listen for bucket change
events, and only if the metadata file and all referenced files are present
would it kick off the analysis.

wtgee, 2018-10-16:
Always happy to close a 2 year old Issue. :) Closed by #687.

closed 2018-10-16 by wtgee