panoptes/POCS #56: Horizon and Obstruction Limits

wtgee, 2016-07-31:
At some point we need to be able to avoid certain sections of the sky or arbitrary horizon limits. At MLO the radio tower ends up in the images for certain targets.

joshwalawender, 2016-08-01:
Originally we had this slated for scheduler.  The idea was that the mount would have horizon limits based on actual, physical collisions and that scheduler would handle sky obstructions.

Our idea was the users could define a single elevation as their horizon, a table of values of horizon as a function of azimuth, or in principle a custom function (if they had some very weird situation).  The scheduler would check that the target would not dip below the horizon during an image.

I think that is still a reasonable strategy.

wtgee, 2016-08-01:
Agreed. We should make an overarching Issue for Scheduling that perhaps
incorporates items we need.

For the weird situations I almost think it would be easiest to mark an
image after the fact such that when you see an image with something
obstructing you could just mark that location.

jamessynge, 2017-04-12:
I'm interested in building, debugging and deploying a Panoptes unit, and I suspect it will help a lot to have an automated approach to identifying the horizon, especially during the process of debugging and site selection. For example, if you could place the tripod, eq mount and camera unit in a location, and have it measure the horizon, this would allow a contributor to find the best location on the available property. Or, during debugging, where you're not aiming to have it running unattended, but rather checking if everything is working well, including if focus and polar alignment are acceptable, having a horizon measurement capability would be useful. This could even be done during the day/evening.

In my case, I'll debug at home, where there are a lot of trees, before taking it to a location (TBD) where a friend or school (again, TBD) will host it. It's possible it'll be moved several times, especially if I take it to a meeting or two for show and tell ;-) before finding a permanent location. I notice from the JPL photos of Panoptes at an open house that at least one prototype was moved to CA, so I think you know what I'm talking about.

wtgee, 2018-01-19:
Happy to close this old issue in favor of #367

closed 2018-01-19 by wtgee