panoptes/POCS #89: POCS Operation.ipynb instructions lead to error

jamessynge, 2017-07-09:
I followed the examples in POCS Operation.ipynb and ran into one of the errors that is causing the tests to fail: 

    AttributeError: 'SkyCoord' object has no attribute 'ndim'

Details:

```
In [13]: new_obs = pocs.observatory.get_observation()
---------------------------------------------------------------------------
AttributeError                            Traceback (most recent call last)
<ipython-input-13-34e5290c9fe4> in <module>()
----> 1 new_obs = pocs.observatory.get_observation()

/var/panoptes/POCS/pocs/observatory.py in get_observation(self, *args, **kwargs)
    159 
    160         self.logger.debug("Getting observation for observatory")
--> 161         self.scheduler.get_observation(*args, **kwargs)
    162 
    163         if self.scheduler.current_observation is None:

/var/panoptes/POCS/pocs/scheduler/dispatch.py in get_observation(self, time, show_all)
     42 
     43         common_properties = {
---> 44             'end_of_night': self.observer.tonight(time=time, horizon=-18 * u.degree)[-1],
     45             'moon': get_moon(time, self.observer.location)
     46         }

/home/panoptes/anaconda3/lib/python3.6/site-packages/astropy/utils/decorators.py in tonight(self, time, horizon, obswl)
    634             name = func.__name__
    635 
--> 636         func = make_function_with_signature(func, name=name, **wrapped_args)
    637         func = functools.update_wrapper(func, wrapped, assigned=assigned,
    638                                         updated=updated)

/home/panoptes/anaconda3/lib/python3.6/site-packages/astropy/units/decorators.py in wrapper(*func_args, **func_kwargs)
    128             # Call the original function with any equivalencies in force.
    129             with add_enabled_equivalencies(self.equivalencies):
--> 130                 return wrapped_function(*func_args, **func_kwargs)
    131 
    132         return wrapper

/home/panoptes/anaconda3/lib/python3.6/site-packages/astroplan/observer.py in tonight(self, time, horizon, obswl)
   1677             if not isinstance(start_time, Time):
   1678                 start_time = Time(start_time)
-> 1679         end_time = self.sun_rise_time(start_time, which='next', horizon=horizon)
   1680 
   1681         return start_time, end_time

/home/panoptes/anaconda3/lib/python3.6/site-packages/astropy/utils/decorators.py in sun_rise_time(self, time, which, horizon)
    634             name = func.__name__
    635 
--> 636         func = make_function_with_signature(func, name=name, **wrapped_args)
    637         func = functools.update_wrapper(func, wrapped, assigned=assigned,
    638                                         updated=updated)

/home/panoptes/anaconda3/lib/python3.6/site-packages/astropy/units/decorators.py in wrapper(*func_args, **func_kwargs)
    128             # Call the original function with any equivalencies in force.
    129             with add_enabled_equivalencies(self.equivalencies):
--> 130                 return wrapped_function(*func_args, **func_kwargs)
    131 
    132         return wrapper

/home/panoptes/anaconda3/lib/python3.6/site-packages/astroplan/observer.py in sun_rise_time(self, time, which, horizon)
   1088         ISO: 2001-02-02 14:02:50.554, JD: 2451943.08531
   1089         """
-> 1090         return self.target_rise_time(time, get_sun(time), which, horizon)
   1091 
   1092     @u.quantity_input(horizon=u.deg)

/home/panoptes/anaconda3/lib/python3.6/site-packages/astropy/utils/decorators.py in target_rise_time(self, time, target, which, horizon)
    634             name = func.__name__
    635 
--> 636         func = make_function_with_signature(func, name=name, **wrapped_args)
    637         func = functools.update_wrapper(func, wrapped, assigned=assigned,
    638                                         updated=updated)

/home/panoptes/anaconda3/lib/python3.6/site-packages/astropy/units/decorators.py in wrapper(*func_args, **func_kwargs)
    128             # Call the original function with any equivalencies in force.
    129             with add_enabled_equivalencies(self.equivalencies):
--> 130                 return wrapped_function(*func_args, **func_kwargs)
    131 
    132         return wrapper

/home/panoptes/anaconda3/lib/python3.6/site-packages/astroplan/observer.py in target_rise_time(self, time, target, which, horizon)
    898                                            dict(time=time, target=target,
    899                                                 which=which, rise_set='rising',
--> 900                                                 horizon=horizon))
    901 
    902     @u.quantity_input(horizon=u.deg)

/home/panoptes/anaconda3/lib/python3.6/site-packages/astroplan/observer.py in _determine_which_event(self, function, args_dict)
    829 
    830         if which == 'next' or which == 'nearest':
--> 831             next_event = function(*args('next'))
    832             if which == 'next':
    833                 return next_event

/home/panoptes/anaconda3/lib/python3.6/site-packages/astroplan/observer.py in _calc_riseset(self, time, target, prev_next, rise_set, horizon, N)
    730             times = _generate_24hr_grid(time, -1, 0, N)
    731 
--> 732         altaz = self.altaz(times, target, grid=True)
    733         altitudes = altaz.alt
    734 

/home/panoptes/anaconda3/lib/python3.6/site-packages/astroplan/observer.py in altaz(self, time, target, obswl, grid)
    464         """
    465         if target is not None:
--> 466             time, target = self._preprocess_inputs(time, target, grid)
    467 
    468         altaz_frame = AltAz(location=self.location, obstime=time,

/home/panoptes/anaconda3/lib/python3.6/site-packages/astroplan/observer.py in _preprocess_inputs(self, time, target, grid)
    396                 if time.isscalar:
    397                     target = target[:, np.newaxis]
--> 398                 while target.ndim <= time.ndim:
    399                     target = target[:, np.newaxis]
    400         if not self._is_broadcastable(target.shape, time.shape):

/home/panoptes/anaconda3/lib/python3.6/site-packages/astropy/coordinates/sky_coordinate.py in __getattr__(self, attr)
    449         # Fail
    450         raise AttributeError("'{0}' object has no attribute '{1}'"
--> 451                              .format(self.__class__.__name__, attr))
    452 
    453     def __setattr__(self, attr, val):

AttributeError: 'SkyCoord' object has no attribute 'ndim'

In [14]: 
```

wtgee, 2017-11-13:
I believe this has been solved by an astropy update. Can you retest and report any additional failures?

wtgee, 2018-01-11:
@jamessynge when you get the change can you check this notebook again. I'd like to close out some old Issues if not a problem.

wtgee, 2018-01-15:
Closing for lack of activity. Please open back up if an issue.

closed 2018-01-15 by wtgee