panoptes/POCS #91: test_moon_veto fails

jamessynge, 2017-07-09:
```
__________________________________________ test_moon_veto ___________________________________________

observer = <Observer: name='Test Observer',
    location (lon, lat, el)=(-155.58 deg, 19.539999999999992 deg, 3400.0000000009422 m),
    timezone=<DstTzInfo 'US/Hawaii' LMT-1 day, 13:29:00 STD>>

    def test_moon_veto(observer):
        mac = MoonAvoidance()
    
        time = Time('2016-08-13 10:00:00')
    
        moon = get_moon(time, observer.location)
    
        observation1 = Observation(Field('Sabik', '17h10m23s -15d43m30s'))  # Sabik
    
        veto1, score1 = mac.get_score(time, observer, observation1, moon=moon)
    
>       assert veto1 is True
E       assert False is True

pocs/tests/test_constraints.py:182: AssertionError
```

wtgee, 2017-08-03:
Fixed and tested in latest commit. Please pull origin. Note upgrade of `astropy` in `requirements.txt`

wtgee, 2017-11-13:
Pinging on the fix along with other Issues related to testing. Please retest and report.

jamessynge, 2017-11-18:
Acknowledged. Will test on laptop (not NUC that has devices on it), along
with writing and testing scripts for installing software.

wtgee, 2017-12-21:
Closing due to inactivity.

closed 2017-12-21 by wtgee