panoptes/POCS #92: test_solve_field_unsolved fails

jamessynge, 2017-07-09:
```
_____________________________________ test_solve_field_unsolved _____________________________________

unsolved_fits_file = '/var/panoptes/POCS/pocs/tests/data/unsolved.fits'

    def test_solve_field_unsolved(unsolved_fits_file):
        im0 = Image(unsolved_fits_file)
    
        assert isinstance(im0, Image)
        assert im0.wcs is None
        assert im0.pointing is None
    
>       im0.solve_field(verbose=True, replace=False, radius=4)

pocs/tests/test_images.py:79: 
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
pocs/images.py:174: in solve_field
    **kwargs)
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _

fname = '/var/panoptes/POCS/pocs/tests/data/unsolved.fits', replace = False, remove_extras = True
kwargs = {'dec': 44.456875, 'ra': 301.56439, 'radius': 4, 'verbose': True}, verbose = True
out_dict = {}
output = 'Reading input file 1 of 1: "/var/panoptes/POCS/pocs/tests/data/unsolved.fits"...\nHeader has 45 cards\nTraceback (mos...ed.nqu93h --outfile /tmp/tmp.ppm.DCF7fX --ppm\n ioutils.c:567:run_command_get_outputs Command failed: return value 1\n'
errs = None, proc = <subprocess.Popen object at 0x7f42807f0710>

    def get_solve_field(fname, replace=True, remove_extras=True, **kwargs):
        """Convenience function to wait for `solve_field` to finish.
    
        This function merely passes the `fname` of the image to be solved along to `solve_field`,
        which returns a subprocess.Popen object. This function then waits for that command
        to complete, populates a dictonary with the EXIF informaiton and returns. This is often
        more useful than the raw `solve_field` function
    
        Args:
            fname ({str}): Name of file to be solved, either a FITS or CR2
            replace (bool, optional): Replace fname the solved file
            remove_extras (bool, optional): Remove the files generated by solver
            **kwargs ({dict}): Options to pass to `solve_field`
    
        Returns:
            dict: Keyword information from the solved field
        """
        verbose = kwargs.get('verbose', False)
        out_dict = {}
        output = None
        errs = None
    
        # Check for solved file
        if kwargs.get('skip_solved', True) and \
                (os.path.exists(fname.replace('.fits', '.solved')) or WCS(fname).is_celestial):
            if verbose:
                print(
                    "Solved file exists, skipping (pass skip_solved=False to solve again): {}".format(fname))
    
            out_dict['solved_fits_file'] = fname
            return out_dict
    
        if verbose:
            print("Entering get_solve_field: {}".format(fname))
    
        # Set a default radius of 15
        kwargs.setdefault('radius', 15)
    
        proc = solve_field(fname, **kwargs)
        try:
            output, errs = proc.communicate(timeout=kwargs.get('timeout', 30))
        except subprocess.TimeoutExpired:
            proc.kill()
            raise error.Timeout("Timeout while solving")
        else:
            if verbose:
                print("Output: {}", output)
                print("Errors: {}", errs)
    
            if not os.path.exists(fname.replace('.fits', '.solved')):
>               raise error.SolveError('File not solved')
E               pocs.utils.error.SolveError: File not solved

pocs/utils/images.py:148: SolveError
--------------------------------------- Captured stdout call ----------------------------------------
Entering get_solve_field: /var/panoptes/POCS/pocs/tests/data/unsolved.fits
Entering solve_field
Cmd:  ['/var/panoptes/POCS/scripts/solve_field.sh', '--guess-scale --cpulimit 15 --no-verify --no-plots --crpix-center --match none --corr none --wcs none --downsample 4 --overwrite --skip-solved --ra 301.56439 --dec 44.456875 --radius 4', '/var/panoptes/POCS/pocs/tests/data/unsolved.fits']
Output: {} Reading input file 1 of 1: "/var/panoptes/POCS/pocs/tests/data/unsolved.fits"...
Header has 45 cards
Traceback (most recent call last):
  File "/usr/lib/python2.7/runpy.py", line 174, in _run_module_as_main
    "__main__", fname, loader, pkg_name)
  File "/usr/lib/python2.7/runpy.py", line 72, in _run_code
    exec code in run_globals
  File "/usr/lib/python2.7/dist-packages/astrometry/util/image2pnm.py", line 323, in <module>
    sys.exit(main())
  File "/usr/lib/python2.7/dist-packages/astrometry/util/image2pnm.py", line 320, in main
    mydir, fix_sdss=options.fix_sdss)
  File "/usr/lib/python2.7/dist-packages/astrometry/util/image2pnm.py", line 248, in convert_image
    (imgtype, errstr) = image2pnm(infile, outfile, sanitized, force_ppm, no_fits2fits, extension, mydir, fix_sdss)
  File "/usr/lib/python2.7/dist-packages/astrometry/util/image2pnm.py", line 184, in image2pnm
    errstr = fits2fits(infile, sanitized, fix_idr=fix_sdss)
  File "/usr/lib/python2.7/dist-packages/astrometry/util/fits2fits.py", line 86, in fits2fits
    hdu.scale('int16', '', bscale, bzero)
  File "/usr/lib/python2.7/dist-packages/astropy/io/fits/hdu/image.py", line 384, in scale
    bzero=bzero, blank=None)
  File "/usr/lib/python2.7/dist-packages/astropy/io/fits/hdu/image.py", line 442, in _scale_internal
    self.data += -_zero
TypeError: Cannot cast ufunc add output from dtype('int32') to dtype('uint16') with casting rule 'same_kind'
augment-xylist.c:588:backtick Failed to run command: /usr/bin/python -m astrometry.util.image2pnm --sanitized-fits-outfile /tmp/tmp.sanitized.r4fbSC --fix-sdss --infile /var/panoptes/POCS/pocs/tests/data/unsolved.fits --uncompressed-outfile /tmp/tmp.uncompressed.nqu93h --outfile /tmp/tmp.ppm.DCF7fX --ppm
 ioutils.c:567:run_command_get_outputs Command failed: return value 1

Errors: {} None
--------------------------------------- Captured stderr call ----------------------------------------
2017-07-09 17:05:26 Panmass panoptes[11800] ERROR SolveError: File not solved

```

wtgee, 2017-11-13:
Maybe related to #111 . Can you test again and report failures here or there and perhaps let's close one of these.

wtgee, 2017-12-21:
Closing due to inactivity.

closed 2017-12-21 by wtgee