panoptes/POCS #93: test_error_exit fails

jamessynge, 2017-07-09:
Looks like the test is looking for the wrong type of error (SystemExit vs. PanError).

```
__________________________________________ test_error_exit __________________________________________

config = {'cameras': {'auto_detect': True, 'devices': [{'model': 'canon_gphoto2'}, {'model': 'canon_gphoto2'}], 'primary': '14d...: <Quantity 3400.0 m>, 'gmt_offset': -600, 'horizon': <Quantity 30.0 deg>, 'latitude': <Quantity 19.54 deg>, ...}, ...}

    def test_error_exit(config):
        with pytest.raises(SystemExit):
>           Observatory(ignore_local_config=True, simulator=['none'])

pocs/tests/test_observatory.py:50: 
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
pocs/observatory.py:50: in __init__
    self._create_cameras(**kwargs)
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _

self = <pocs.observatory.Observatory object at 0x7f42b8264470>
kwargs = {'ignore_local_config': True, 'simulator': ['none']}
camera_info = {'auto_detect': True, 'devices': [{'model': 'canon_gphoto2'}, {'model': 'canon_gphoto2'}], 'primary': '14d3bd'}
a_simulator = False, ports = [], auto_detect = True

    def _create_cameras(self, **kwargs):
        """Creates a camera object(s)
    
            Loads the cameras via the configuration.
    
            Creates a camera for each camera item listed in the config. Ensures the
            appropriate camera module is loaded.
    
            Note: We are currently only operating with one camera and the `take_pic.sh`
                script automatically discovers the ports.
    
            Note:
                This does not actually make a usb connection to the camera. To do so,
                call the 'camear.connect()' explicitly.
    
            Args:
                **kwargs (dict): Can pass a camera_config object that overrides the info in
                    the configuration file. Can also pass `auto_detect`(bool) to try and
                    automatically discover the ports.
    
            Returns:
                list: A list of created camera objects.
    
            Raises:
                error.CameraNotFound: Description
                error.PanError: Description
            """
        if kwargs.get('camera_info') is None:
            camera_info = self.config.get('cameras')
    
        self.logger.debug("Camera config: \n {}".format(camera_info))
    
        a_simulator = 'camera' in self.config.get('simulator', [])
        if a_simulator:
            self.logger.debug("Using simulator for camera")
    
        ports = list()
    
        # Lookup the connected ports if not using a simulator
        auto_detect = kwargs.get(
            'auto_detect', camera_info.get('auto_detect', False))
        if not a_simulator and auto_detect:
            self.logger.debug("Auto-detecting ports for cameras")
            try:
                ports = list_connected_cameras()
            except Exception as e:
                self.logger.warning(e)
    
            if len(ports) == 0:
                raise error.PanError(
>                   msg="No cameras detected. Use --simulator=camera for simulator.")
E               pocs.utils.error.PanError

pocs/observatory.py:589: PanError
--------------------------------------- Captured stderr call ----------------------------------------
2017-07-09 17:05:34 Panmass panoptes[11800] INFO Initializing observatory
2017-07-09 17:05:34 Panmass panoptes[11800] INFO 	Setting up location
2017-07-09 17:05:34 Panmass panoptes[11800] INFO 	Setting up mount
2017-07-09 17:05:34 Panmass panoptes[11800] INFO Loading mount commands file: /var/panoptes/POCS/resources/mounts/ioptron.yaml
2017-07-09 17:05:36 Panmass panoptes[11800] INFO Creating iOptron mount
2017-07-09 17:05:36 Panmass panoptes[11800] INFO Mount created
2017-07-09 17:05:36 Panmass panoptes[11800] INFO 	Setting up cameras
2017-07-09 17:05:37 Panmass panoptes[11800] ERROR PanError: No cameras detected. Use --simulator=camera for simulator.
```

wtgee, 2017-11-13:
Some of the testing suite has been fixed since this time. Can you run again and report if still failing.

wtgee, 2017-11-16:
I think this relates to having a camera physically connected while running the test. Some work needs to be done to respect the `--hardware` or `--camera` flag. In general need to make the documentation clear about how to run the test suite to test the software vs how to run it to test the hardware.

wtgee, 2017-12-21:
I think this has been resolved. Closing due to inactivity.

closed 2017-12-21 by wtgee