panoptes/POCS #94: Add instructions for creating the "panoptes" user

jamessynge, 2017-07-09:
We need explicit/detailed instructions for setting up the NUC, including creating the "panoptes" user. In particular, info about adding that user to the dialout group (else it has no access to the Arduino's and the USB-to-Serial adapters).
Also info about creating the /var/panoptes directory (must create the astrometry/data subdir, else some tests fail).

jamessynge, 2017-11-03:
New info: the panoptes user needs to be in the plugdev group, else has no access to the cameras when not logged in via a locally attached monitor & keyboard, including IIUC via the auto-started tmux session.

wtgee, 2020-02-16:
Closed by #951.  There are other things that need to be done as part of that process, mostly documentation, but new specific Issues should be opened to identify those.

PR #859 (2019-10-13): Utils and docker
**Breaking Changes**

This will move POCS over to a Docker based system, utilizing the panoptes-utils module

Will replace #851 and #778 

## Description

## Related Issue
#851 #778 #850 #734 

Closes #531 
Closes #519 
Closes #238 
Closes #555 
Closes #137 
Closes #94
closed 2020-02-16 by wtgee