panoptes/POCS #96: Log to unique filename by program and time

jamessynge, 2017-08-06:
Running peas_shell logs to panoptes.log, as does pocs_shell. They should each log to a unique file name.
And it would help debugging if each run created a new file, e.g. with a timestamp in the name.
To make it easier to find the latest log file, create a symlink from <progname>.log to <progname>.<timestamp>.log.

wtgee, 2017-11-13:
I'm assuming last sentence just means to symlink the timestamped version to `panoptes.log`.

I think with the consolidation of the repos in #107 we might eventually be getting rid of peas_shell. Regardless, I like having the unified log for the two although don't feel too strongly about it. If we keep them separate we should rename to `pocs.log` and `peas.log`.

jamessynge, 2017-11-25:
I'm planning to work on this soon, where I intend to take an approach based on Google's c++ logger: each execution of program A creates a unique file name containing the name of the program, A, the process id and a timestamp. It also creates or overwrites a symlink A.log to that unique name.

wtgee, 2017-12-31:
Closed in #145 and release [v0.6.0](https://github.com/panoptes/POCS/releases/tag/v0.6.0)

PR #145 (2017-11-30): Log file updates
* Support use of `{}` in logger statements so we can avoid superfluous `format` calls
* Update CONTRIBUTING doc for logging
* Update log file name (#96)

Will probably need to do something in the future to clean up old log files.

Long-running (i.e. multi-day) programs will still be rotated daily at 11:30 am local. These will carry an additional day stamp.

Generates log files with current PID and a utc timestamp:
```
$ ls /var/panoptes/logs
panoptes.log -> /var/panoptes/logs/panoptes-26947-20171128T035245Z.log
panoptes-26947-20171128T035245Z.log
```

Note that this is being m [... cut]
commit 6aa08546 (2017-11-30): Log file updates (#145)

* * Making the logger use the python3 `format` placeholder
* Small change to test above
* Add info about logging to CONTRIBUTING file

* Log to unique filename and symlink to generic

Generates log files with current PID and a utc timestamp:

```
panoptes.log -> /var/panoptes/logs/panoptes-26947-20171128T035245Z.log
panoptes-26947-20171128T035245Z.log
```

Closes #96

* More explicit logging examples

* Change example to conform to logging

* * Make a PanLogger that is a thin-wrapper around some of the logging
messages but allows for `format` style
* Update Contributin [... cut]
closed 2017-12-31 by wtgee