panoptes/POCS #103: PanMongo.insert_current fails to insert/replace some records

jamessynge, 2017-11-04:
peas_shell emits these errors periodically:

`/var/panoptes/POCS/pocs/utils/database.py:84: UserWarning: Problem inserting object into collection: The dotted field '4.16, "z":-8.27, "o"' in 'data.camera_board.4.16, "z":-8.27, "o"' is not valid for storage.
 warn("Problem inserting object into collection: {}".format(e))
/var/panoptes/POCS/pocs/utils/database.py:84: UserWarning: Problem inserting object into collection: The dotted field '3, "z":-8.03, "o"' in 'data.camera_board.3, "z":-8.03, "o"' is not valid for storage.
  warn("Problem inserting object into collection: {}".format(e))
/var/panoptes/POCS/pocs/utils/database.py:84: UserWarning: Problem inserting object into collection: The dotted field '24, "z":-7.93, "o"' in 'data.camera_board.24, "z":-7.93, "o"' is not valid for storage.
  warn("Problem inserting object into collection: {}".format(e))
/var/panoptes/POCS/pocs/utils/database.py:84: UserWarning: Problem inserting object into collection: The dotted field '23, "z":-7.92, "o"' in 'data.camera_board.23, "z":-7.92, "o"' is not valid for storage.
  warn("Problem inserting object into collection: {}".format(e))
/var/panoptes/POCS/pocs/utils/database.py:84: UserWarning: Problem inserting object into collection: The dotted field '18, "z":-8.44, "o"' in 'data.camera_board.18, "z":-8.44, "o"' is not valid for storage.
  warn("Problem inserting object into collection: {}".format(e))
`

Here is a SO page about the problem and a potential solution: https://stackoverflow.com/questions/30014243/mongoerror-the-dotted-field-is-not-valid-for-storage

wtgee, 2017-11-04:
Actually, I think this is a problem when the full string coming off the arduino isn't a valid json string. It then sometimes parses wrong and turns what should be values into keys and then tries to insert, generating this error. We don't (i.e. shouldn't) have any keys that have dots in them.

It's usually not a valid json because the buffering on the serial has gone wrong, something I was hoping your updated code would fix. I would first check for a full valid json before looking into above solution.

jamessynge, 2017-12-27:
I also see this happen when running test_pocs.py:

pocs/tests/test_pocs.py::test_run
  /home/james/git/panoptes/POCS/pocs/utils/database.py:100: UserWarning: Problem inserting object into collection: Cannot encode object: <Time object: scale='utc' format='jyear_str' value=J2000.000>
    warn("Problem inserting object into collection: {}".format(e))

wtgee, 2017-12-27:
The time type should be handled by the BSON encoder although I had seen the same problems recently with https://github.com/panoptes/POCS/pull/225 (which was fixed by explicitly converting that one value away from an astropy.time.Time` object). Is this the same type of time or is it some derivative of `datetime`? 

The fix for this should be the same as for https://github.com/panoptes/POCS/issues/206, that is, we should have one way to scrub objects whether they are going into mongo or out as a message.

jamessynge, 2017-12-27:
Looks to be an astropy type (based on the jyear_str and J2000.000, and a Google search for that text).

wtgee, 2018-01-11:
I'm still seeing this intermittently on PAN001 with new camera board and arduino shared libraries.

jamessynge, 2018-01-12:
FWIW, I'm skeptical that the cause is invalid JSON from the boards. I think
the key used when updating a doc in the db is somehow corrupt as far as
mongo is concerned

wtgee, 2018-01-12:
This is what I'm seeing frequently on PAN001 right now:
```
warn("Problem inserting object into collection: {}".format(e))
/var/panoptes/POCS/pocs/utils/database.py:100: UserWarning: Problem inserting object into collection: The dotted field '0.99, "z":-0.02, "o"' in 'data.camera_board.0.99, "z":-0.02, "o"' is not valid for storage.
  warn("Problem inserting object into collection: {}".format(e))
/var/panoptes/POCS/pocs/utils/database.py:100: UserWarning: Problem inserting object into collection: The dotted field '7, "y":0.99, "z":-0.02, "o"' in 'data.camera_board.7, "y":0.99, "z":-0.02, "o"' is not valid for storage.
```

Seems to be some bad string concatenation around `camera_board`. The line inserting it is:

```
            current_obj = {
                'type': collection,
                'data': obj,
                'date': current_time(datetime=True),
            }
```

which is where the `data` comes from in above warning.

jamessynge, 2018-01-26:
Another insert failure, but now with the record that couldn't be inserted:

`
PEAS > /var/panoptes/POCS/pocs/utils/database.py:88: UserWarning: Problem inserting object into collection: MongoDB can only handle up to 8-byte ints, {'type': 'environment', 'data': {'telemetry_board': {'name': 'telemetry_board', 'millis': 980784041, 'report_num': 417104, 'ver': '2018-01-14', 'power': {'computer': 1, 'fan': 1, 'mount': 1, 'cameras': 1, 'weather': 1, 'main': 1}, 'current': {'main': 421, 'fan': 49, 'mount': 32, 'cameras': 181}, 'amps': {'main': 1178.8, 'fan': 88.2, 'mount': 57.6, 'cameras': 181.0}, 'humidity': 13.5, 'temp_00': -1.1, 'temperature': [-4.31, -4.63, 0.69], 'date': '2018-01-26T11:48:59 GMT'}, 'camera_board': {'name': 'camera_board', 'count': 27674957142767538857, 'num': 27513, 'inputs': 9, 'camera_00': 1, 'camera_01': 1, 'accelerometer': {'x': -6.96, 'y': 6.86, 'z': 1.57, 'o': 7}, 'humidity': 29.5, 'temp_00': -6.0, 'date': '2018-01-26T11:48:59 GMT'}}, 'date': datetime.datetime(2018, 1, 26, 11, 48, 59, 444352)}
`

wtgee, 2018-01-26:
`'count': 27674957142767538857` isn't very helpful. :) Not sure how it gets that high though as you are doing a subtraction [here](https://github.com/panoptes/POCS/blob/develop/resources/arduino_files/camera_board/camera_board.ino#L167) (although there is still a TODO there).

jamessynge, 2018-01-27:
Yeah, I was sure that count was totally bogus. I don't think that value can be produced by the sketch, so it may mean that the reading the sketch output failed in some way.

wtgee, 2018-07-26:
Just a bump as I started getting this problem again on PAN001. We haven't updated the camera board software in quite some time but seems due:

```
/var/panoptes/POCS/pocs/utils/database.py:33: UserWarning: Problem inserting object
 into collection: MongoDB can only handle up to 8-byte ints

# snip

 'count': 35177157493517757885, 
```

wtgee, 2019-06-29:
This should be solved with the merge up `panoptes-utils` release above.

closed 2019-06-29 by wtgee