panoptes/POCS #110: pocs/utils/data.py crashes

jamessynge, 2017-11-11:
`panoptes@Panmass:/var/panoptes/POCS$ python pocs/utils/data.py
Traceback (most recent call last):
  File "pocs/utils/data.py", line 5, in <module>
    from astroplan import download_IERS_A
  File "/home/panoptes/anaconda3/lib/python3.6/site-packages/astroplan/__init__.py", line 18, in <module>
    from ._astropy_init import *
  File "/home/panoptes/anaconda3/lib/python3.6/site-packages/astroplan/_astropy_init.py", line 116, in <module>
    from astropy import config
  File "/home/panoptes/anaconda3/lib/python3.6/site-packages/astropy/__init__.py", line 116, in <module>
    _check_numpy()
  File "/home/panoptes/anaconda3/lib/python3.6/site-packages/astropy/__init__.py", line 104, in _check_numpy
    from .utils import minversion
  File "/home/panoptes/anaconda3/lib/python3.6/site-packages/astropy/utils/__init__.py", line 20, in <module>
    from .compat.odict import OrderedDict
  File "/home/panoptes/anaconda3/lib/python3.6/site-packages/astropy/utils/compat/__init__.py", line 14, in <module>
    from .numpycompat import *
  File "/home/panoptes/anaconda3/lib/python3.6/site-packages/astropy/utils/compat/numpycompat.py", line 18, in <module>
    NUMPY_LT_1_8 = not minversion('numpy', '1.8.0')
  File "/home/panoptes/anaconda3/lib/python3.6/site-packages/astropy/utils/introspection.py", line 155, in minversion
    from pkg_resources import parse_version
  File "<frozen importlib._bootstrap>", line 961, in _find_and_load
  File "<frozen importlib._bootstrap>", line 950, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 646, in _load_unlocked
  File "<frozen importlib._bootstrap>", line 616, in _load_backward_compatible
  File "/home/panoptes/anaconda3/lib/python3.6/site-packages/setuptools-27.2.0-py3.6.egg/pkg_resources/__init__.py", line 2985, in <module>
  File "/home/panoptes/anaconda3/lib/python3.6/site-packages/setuptools-27.2.0-py3.6.egg/pkg_resources/__init__.py", line 2971, in _call_aside
  File "/home/panoptes/anaconda3/lib/python3.6/site-packages/setuptools-27.2.0-py3.6.egg/pkg_resources/__init__.py", line 3013, in _initialize_master_working_set
  File "/home/panoptes/anaconda3/lib/python3.6/site-packages/setuptools-27.2.0-py3.6.egg/pkg_resources/__init__.py", line 2544, in activate
  File "/home/panoptes/anaconda3/lib/python3.6/site-packages/setuptools-27.2.0-py3.6.egg/pkg_resources/__init__.py", line 2118, in declare_namespace
  File "/home/panoptes/anaconda3/lib/python3.6/site-packages/setuptools-27.2.0-py3.6.egg/pkg_resources/__init__.py", line 2058, in _handle_ns
  File "/home/panoptes/anaconda3/lib/python3.6/site-packages/setuptools-27.2.0-py3.6.egg/pkg_resources/__init__.py", line 2087, in _rebuild_mod_path
AttributeError: '_NamespacePath' object has no attribute 'sort'
panoptes@Panmass:/var/panoptes/POCS$ 
`

jamessynge, 2017-11-11:
```
panoptes@Panmass:/var/panoptes/POCS$ python pocs/utils/data.py
Traceback (most recent call last):
  File "pocs/utils/data.py", line 5, in <module>
    from astroplan import download_IERS_A
  File "/home/panoptes/anaconda3/lib/python3.6/site-packages/astroplan/__init__.py", line 18, in <module>
    from ._astropy_init import *
  File "/home/panoptes/anaconda3/lib/python3.6/site-packages/astroplan/_astropy_init.py", line 116, in <module>
    from astropy import config
  File "/home/panoptes/anaconda3/lib/python3.6/site-packages/astropy/__init__.py", line 116, in <module>
    _check_numpy()
  File "/home/panoptes/anaconda3/lib/python3.6/site-packages/astropy/__init__.py", line 104, in _check_numpy
    from .utils import minversion
  File "/home/panoptes/anaconda3/lib/python3.6/site-packages/astropy/utils/__init__.py", line 20, in <module>
    from .compat.odict import OrderedDict
  File "/home/panoptes/anaconda3/lib/python3.6/site-packages/astropy/utils/compat/__init__.py", line 14, in <module>
    from .numpycompat import *
  File "/home/panoptes/anaconda3/lib/python3.6/site-packages/astropy/utils/compat/numpycompat.py", line 18, in <module>
    NUMPY_LT_1_8 = not minversion('numpy', '1.8.0')
  File "/home/panoptes/anaconda3/lib/python3.6/site-packages/astropy/utils/introspection.py", line 155, in minversion
    from pkg_resources import parse_version
  File "<frozen importlib._bootstrap>", line 961, in _find_and_load
  File "<frozen importlib._bootstrap>", line 950, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 646, in _load_unlocked
  File "<frozen importlib._bootstrap>", line 616, in _load_backward_compatible
  File "/home/panoptes/anaconda3/lib/python3.6/site-packages/setuptools-27.2.0-py3.6.egg/pkg_resources/__init__.py", line 2985, in <module>
  File "/home/panoptes/anaconda3/lib/python3.6/site-packages/setuptools-27.2.0-py3.6.egg/pkg_resources/__init__.py", line 2971, in _call_aside
  File "/home/panoptes/anaconda3/lib/python3.6/site-packages/setuptools-27.2.0-py3.6.egg/pkg_resources/__init__.py", line 3013, in _initialize_master_working_set
  File "/home/panoptes/anaconda3/lib/python3.6/site-packages/setuptools-27.2.0-py3.6.egg/pkg_resources/__init__.py", line 2544, in activate
  File "/home/panoptes/anaconda3/lib/python3.6/site-packages/setuptools-27.2.0-py3.6.egg/pkg_resources/__init__.py", line 2118, in declare_namespace
  File "/home/panoptes/anaconda3/lib/python3.6/site-packages/setuptools-27.2.0-py3.6.egg/pkg_resources/__init__.py", line 2058, in _handle_ns
  File "/home/panoptes/anaconda3/lib/python3.6/site-packages/setuptools-27.2.0-py3.6.egg/pkg_resources/__init__.py", line 2087, in _rebuild_mod_path
AttributeError: '_NamespacePath' object has no attribute 'sort'
panoptes@Panmass:/var/panoptes/POCS$ 
```

wtgee, 2017-11-11:
Haven't seen that before. @jamessynge can you list your astropy and astroplan versions? This looks like a bug in astroplan as it doesn't actually hit any of our code.

wtgee, 2017-12-21:
Closing due to inactivity.

closed 2017-12-21 by wtgee