panoptes/POCS #118: Encourage or require the use of a source formatter

jamessynge, 2017-11-15:
It would be helpful if we had a consistent style for code. I like the golang approach: they provide gofmt, a tool that is applied to all "official" code, and submit fails if the code doesn't exactly match the format that gofmt would produce for the input file. This tool (https://github.com/google/yapf) aims to be used the same way for python. I use a variant of it at work, and like the fact that when I'm editing I don't have to spend time thinking about formatting because I just run the formatter periodically and before sending for review.

wtgee, 2017-11-15:
We do require standard pep8 in the sense that our testing suite will fail if it doesn't conform. This is enforced via `pocs/tests/test_codestyle.py`. 

The only part we don't conform to is E501, line length of 80 chars, mostly because that drives me bonkers. I think 120 is a more sane default for people these days (I try to keep docstrings at 80).  We also don't conform on #402 (all import statements at top) because of a few places we do some trickery. Probably better to just ignore those files.

I think this is mostly a documentation issue about how to run tests before committing and how to setup an editor to do this automatically.

wtgee, 2017-11-15:
That said, I just noticed the test file only checks the `pocs` directory, so we should make note to expand that to `peas` in #98

jamessynge, 2017-11-15:
While typing it is easier to go with wide line length, but when running diff, 120 becomes a bit too much.

joshwalawender, 2017-11-15:
I used to be more strongly offended by the 80 character limit, but I've learned to live with it on other projects.  It does seem silly to have software written on 4k and 5k monitors to be limited by the number of characters that could fit on a punch card ...

jamessynge, 2017-11-18:
My laptop isn't 4K or 5K wide, and my eyes aren't what they were when I
used to complain about 80 column limits.
There is reliably enough room on a single screen for two 80 or 100 column
wide editors, but 120 is usually pushing it.
For example, I just made a maximum width gnome terminal on my laptop, and
it is 204 columns wide.
I usually use Meld for graphical diffs, and with gutters and two
scrollbars, 100 columns would actually be too wide.
And it is fairly easy to apply these tools automatically using a git
pre-commit hook, so you don't need to think about it while editing a file.

So, I repeat my request: Let's make it a policy to keep the the maximum
width that of a punch card, or 90 if you prefer a bit more room.

wtgee, 2017-11-18:
I've held out on this for a long time but that was easy since I was doing most of the development. :)

If we go less than 120 then we should conform to the 80 and just keep in line with pep.

This should perhaps be done as part of #107

wtgee, 2017-11-18:
Note that pep8 is actually 79 characters for code and 72 for docstrings

https://www.python.org/dev/peps/pep-0008/#maximum-line-length

Note also they "allow" 100 over 80 if strongly preferred. Again, I could make a case but don't necessarily think we should go back and forth on this. I'm ~~happy~~ willing to switch to 80 without issue.

jamessynge, 2017-11-19:
I'll make use of yapf on my current change, and we can get a feel for it. It uses pep8 as the style by default.

wtgee, 2017-12-05:
Is this okay to close?

closed 2017-12-05 by jamessynge