panoptes/POCS #125: Store telemetry and camera board data in different docs/dbs

jamessynge, 2017-11-19:
Since the data from these two boards comes arrives in an uncoordinated fashion, we should simply record the data when it arrives, and not try to tie them together into a single record.

wtgee, 2017-11-19:
Agreed. This involves work fixing up `peas/sensors.py` so that it doesn't listen for two devices at once but each device as own instance of class.

jamessynge, 2018-01-10:
On PAN006, the telemetry board is not responding, so the lack of separation of telemetry and camera readings is blocking setup_pocs from working.

wtgee, 2018-01-10:
It sounds like it is hanging while trying to read?

jamessynge, 2018-01-10:
That's what I assumed, though I now think there must be multiple issues:

- setup_pocs is encountering a timeout, which I assume is from attempting to read the latest environment data from PEAS.
- peas_shell is not reporting any values for `last_reading environment` so either I'm specifying the wrong arg to the command or there is a problem with collecting that environment data.

There are no working tests of peas, let along peas_shell, so we could have introduced multiple problems with our changes to pocs/utils, but be unaware.

wtgee, 2018-01-11:
Do you have mongo set up correctly? It occurred to me the `last_reading is reading from mongo direct.

jamessynge, 2018-01-11:
So far as I know, mongo is correct. For example, `last_reading weather`
works.

As we've discussed elsewhere, there are bugs in peas/sensor.py; my fault as
I fixed things, but we didn't have good tests of that code.

wtgee, 2018-02-13:
Closing in favor of #420

closed 2018-02-13 by wtgee