panoptes/POCS #128: Tests leave new files in pocs/tests/data

jamessynge, 2017-11-24:
According to `git status`:

	Untracked files:
	(use "git add <file>..." to include in what will be committed)

	pocs/tests/data/unsolved-indx.xyls
	pocs/tests/data/unsolved.axy
	pocs/tests/data/unsolved.new
	pocs/tests/data/unsolved.rdls
	pocs/tests/data/unsolved.solved

We should use a temp directory for output.

wtgee, 2017-11-25:
A temp dir is not a bad idea but these files should have been cleaned up by various methods. Actually, the file should not have been solved in the first place. This is a really small image that shouldn't be solvable by our system.

wtgee, 2017-12-21:
@jamessynge pinging on this again.

wtgee, 2017-12-31:
Closing due to inactivity. Feel free to open if still an issue.

closed 2017-12-31 by wtgee