panoptes/POCS #130: Consider reading temperature from CPU, using to control computer box fan

jamessynge, 2017-11-24:
We have a fan in the computer box, but don't automatically turn it on or off, but we really should make better use of that hardware.

We have temp sensors in the computer box. And the CPU/GPU/motherboard expose temperature sensors (see the Linux package lm-sensors). Any of these might be used to decide when to turn the fan on or off.

Given that not all models of the NUC have fans, and that we have sensors in the case, we might use choose to have two triggers: PEAS could monitor the temperatures in the computer box and of the NUC, and if any of them rise too high it could turn on the fan.

kmeagle1515, 2019-03-25:
HELLO!
Is this related to **GKRELLM**

AnthonyHorton, 2019-03-26:
@kmeagle1515 No. GKrellM is a graphical system monitor application. This Issue is about adding functionality to the POCS/PEAS software to automatically control fans in response to temperature sensor readings.

kmeagle1515, 2019-03-26:
Internal Arduino Temperature Sensor, this should do the work i suppose.
Any thoughts on it?

AnthonyHorton, 2019-03-26:
These are questions for @wtgee & @jamessynge. I'm not very familiar with the PANOPTES hardware myself. I work on the [Huntsman Telescope](https://huntsman.space).

wtgee, 2019-03-26:
@kmeagle1515 thanks for the interest in the issue. @jamessynge is actually talking about two different systems for tuning on the fan:
1. Based on the internal CPU temperature. Something like the `lm-sensors` could be used here, however it's just as easy to use `psutil`. Here's are reading from `PAN001` right now:
```python
In [1]: import psutil                                                                                                                                                                                            

In [2]: psutil.sensors_temperatures()                                                                                                                                                                            
Out[2]: 
{'acpitz': [shwtemp(label='', current=27.8, high=105.0, critical=105.0),
  shwtemp(label='', current=29.8, high=105.0, critical=105.0)],
 'coretemp': [shwtemp(label='Package id 0', current=30.0, high=100.0, critical=100.0),
  shwtemp(label='Core 0', current=27.0, high=100.0, critical=100.0),
  shwtemp(label='Core 1', current=30.0, high=100.0, critical=100.0)],
 'nct6776': [shwtemp(label='SYSTIN', current=123.0, high=0.0, critical=None),
  shwtemp(label='PCH_CPU_TEMP', current=0.0, high=None, critical=None),
  shwtemp(label='CPUTIN', current=20.5, high=80.0, critical=80.0),
  shwtemp(label='AUXTIN', current=25.0, high=80.0, critical=80.0),
  shwtemp(label='PECI Agent 0', current=28.5, high=80.0, critical=100.0),
  shwtemp(label='PCH_CHIP_TEMP', current=0.0, high=80.0, critical=80.0),
  shwtemp(label='PECI Agent 1', current=0.0, high=80.0, critical=80.0)]}
```
2. Based on the temperature sensors that we have installed in the [control box](https://projectpanoptes.org/articles/panoptes-baseline-unit-description/). Here is there current reading (only showing relevant portion):
```python
PEAS > last_reading environment
********************************************************************************
ENVIRONMENT:
{
 'data': {'camera_board': {
                           'date': '2019-03-26T05:56:05 GMT',
                           'humidity': 1.0,
                           'name': 'telemetry_board',
                           'temp_00': 11.1,
                           'temperature': [18.5, 8.5, 5.31],
}},
 'type': 'environment'}
********************************************************************************
39.4 seconds old

PEAS > last_reading weather
********************************************************************************
WEATHER:
{
 'data': {'ambient_temp_C': 3.15,
          'date': datetime.datetime(2019, 3, 26, 5, 56, 25, 682000, tzinfo=<bson.tz_util.FixedOffset object at 0x7f0f08876898>),
          'gust_condition': 'Calm',
          'rain_condition': 'Dry',
          'rain_frequency': 2688.0,
          'rain_sensor_temp_C': '7.34',
          'safe': True,
          'sky_condition': 'Clear',
          'sky_temp_C': -36.37,
          'wind_condition': 'Calm',
          'wind_speed_KPH': 0.0},
 'type': 'weather'}
********************************************************************************
64.6 seconds old
```

However there are a couple of parts to this Issue.  Namely:

* [ ] Decide on temperature cutoff limits and add to config.
* [ ] Write a script that checks the temperature and turns the fan off and on accordingly. This script will use the `pustil` module to get the system temperature. For other temperatures you can either:
  * [ ] Listen to zeromq status messages for the `weather` and the `environment` (preferred).
  * [ ] Read the `weather` and `environment` records from the database (actually defaults to flat json files rather than a real database).
* [ ] Run this script as a separate process from within the [PEAS shell](https://github.com/panoptes/POCS/blob/develop/bin/peas_shell).

Each of those above parts will have various details to them but let's start with that. Each check box would ideally be turned into (at least) one separate Issue/PR.

kmeagle1515, 2019-03-27:
Hello!
@wtgee where will i get temperature threshold values for the computer box used here.

wtgee, 2019-03-27:
The threshold values should be read from the `peas_conf` config file under
some appropriate entry. Part of this PR would be the creation of those
initial  entries.

wtgee, 2024-03-28:
Closing as stale for now as much of this has changed.