panoptes/POCS #137: Create config yaml file(s) for use by tests

jamessynge, 2017-11-26:
I want a config file that can be used by the dome tests, and it makes sense to have another such config file for the bisque tests.

The existing POCS/conf_files/pocs.yaml seems to really be the local file for PAN001. Perhaps we need to have some other file as the base yaml file?

Regarding tests, we could use a separate file such as pocs_test.yaml, or even multiple such files so that we can test the effect of having different numbers of cameras, or different state machines. I can't add a dome section to pocs.yaml without it affecting PAN001 and PAN006.

wtgee, 2017-11-26:
Sorry, this is mostly a documentation issue.

`conf_files.pocs.yaml` is the default config file that should be used for tests and to verify the software runs. It's mostly meant as an example. Any file in the `$POCS/conf_file/` directory can have a `foo_local.yaml` version that will override and add to the `foo.yaml` file. `pocs.yaml` should never be altered, which is why it is part of the repository. A local unit should therefore have a `$POCS/conf_files/pocs_local.yaml` file for any local-specific settings.

wtgee, 2020-02-16:
Will be closed by #951.

PR #859 (2019-10-13): Utils and docker
**Breaking Changes**

This will move POCS over to a Docker based system, utilizing the panoptes-utils module

Will replace #851 and #778 

## Description

## Related Issue
#851 #778 #850 #734 

Closes #531 
Closes #519 
Closes #238 
Closes #555 
Closes #137 
Closes #94
PR #951 (2020-06-01): Docker into develop 02
Merging the `docker` branch into `develop`.

This PR is effectively `v0.7.0`

## Related Issue
Replaces #950 
Closes #927 
Closes #934 
Closes #948 
Closes #531 
Closes #137
Closes #734 
Closes #960
Closes #828 
Closes #952 
Closes #897 
Closes #809 
Closes #970 
Related to #115 

Relies on https://github.com/panoptes/panoptes-utils/pull/189

## Types of changes

- [x] New feature (non-breaking change which adds functionality)
- [x] Breaking change (fix or feature that would cause existing functionality to change)
closed 2020-06-01 by wtgee