panoptes/POCS #174: Use root relative imports, not file or directory relative imports.

jamessynge, 2017-12-03:
Per our [discussion](https://github.com/orgs/panoptes/teams/pocs-developers/discussions/3), we will use imports starting with pocs or peas, and will import packages and modules but not variables, classes and functions. For example:

`import pocs`
`from pocs.utils import images`

To avoid destroying our productivity, we should do this incrementally, perhaps as we modify files. In a while, we can go back and fix missed files.

jamessynge, 2017-12-03:
We should start with modifying CONTRIBUTING.md to describe this.

wtgee, 2017-12-27:
#252 took care of most of these so I'm closing this issue. While we want to remain vigilant about making sure we do this going forward I'm not sure I want the Issue hanging around. Closing for now.

closed 2017-12-27 by wtgee