panoptes/POCS #181: Support pyserial options in yaml files

jamessynge, 2017-12-09:
PySerial supports getting all properties as a dictionary, and replacing some or all properties with a dictionary. I'm proposing that we provide the ability to supply those properties via the yaml file. For example, instead of this (from `conf_files/pocs.yaml`):

```
mount:
    brand: ioptron
    model: 30
    driver: ioptron
    port: /dev/ttyUSB0
    non_sidereal_available: True
```

We'd instead have:
```
mount:
    brand: ioptron
    model: 30
    driver: ioptron
    non_sidereal_available: True
    serial:
       port: /dev/ttyUSB0
       baudrate: 9600
       timeout: 2.0
```

Besides adding the ability to set all of the PySerial properties, this has the added benefit of addressing the pylint complaint that the SerialData constructor takes too many arguments.

wtgee, 2017-12-10:
Yes, I like this more.

jamessynge, 2017-12-26:
I note that Observatory.initialize is supposed to be responsible for setting up the communication with devices. I'm thinking we can extend the serial section with info about what to do when the rs232 device is created. For example:
```yaml
mount:
    brand: ioptron
    model: 30
    driver: ioptron
    non_sidereal_available: True
    serial:
       port: /dev/ttyUSB0
       baudrate: 9600
       timeout: 2.0
       validate_port: True   # If true, then the port must exist at create time.
       open_port: False      # If true, then opens port immediately.
```

`open_port: True` implies `validate_port: True` simply because the port must exist in order to be opened. And since Observatory.initialize is supposed to setup communication, I'm proposing that `open_port` be `False` by default, but that `validate_port` be `True` by default.

I propose to create method `rs232.create_serial_data_from_config(serial_config, name)`, where name is something like 'mount' or 'ioptron', determined by the caller. The serial_config param is just the contents of the serial section in the config. This `rs232.create_serial_data_from_config` will be called by `dome.create_dome_from_config`, before creating the Dome instance.

@wtgee Does this make sense to you?

wtgee, 2017-12-27:
Yes, sounds good and makes sense, but is seems like the caller would already be inside the `mount` or `ioptron` so the `name` param would be redundant. It seems like `create_serial_data_from_config` shouldn't really care what it is creating a connection for, it should just return and create connection. The caller should be responsible for passing it the relevant portion of the config. Unless you had something else in mind that happens inside `create_serial_data_from_config`. 

Does `create_serial_connection_from_config` make more sense despite the verbosity. Maybe since it is in `rs232` it would just be `rs232.create_connection_from_config`

jamessynge, 2018-01-18:
Perhaps we should remove the default baud rate from SerialData, as it isn't possible to choose a good default. For the Arduino Micros it *may* not make a difference as the data never goes over an RS 232 line... Not sure.

jamessynge, 2018-09-09:
The task in the title is complete AFAICT, but the addition of a `create_serial_connection_from_config` hasn't happened. I'm going to close this, and we can create another issue if desired later.

closed 2018-09-09 by jamessynge