panoptes/POCS #193: PanBase creates a PanMongo instance for each created sub-class

jamessynge, 2017-12-14:
This seems likely to be a bad idea.

wtgee, 2017-12-14:
My understanding is that pymongo creates a really light-weight wrapper for the connection and that it is not a problem to have multiple. Worth checking though.

jamessynge, 2017-12-20:
I've reviewed the documentation, and it appears that you need only a single MongoClient instance for each python process, as each instance provides connection pooling. I think this explains the large number of mongo related threads I've seen in pytest timeout thread dumps.

I'm not clear though on how fixing this would interact with multiprocessing. Since we're regularly creating PanMongo instances, then we have separate instances in the sub-processes, which avoids any cross-process sharing problems. But if we have a single instance in the POCS process, what happens in the sub-processes? According to [stackoverflow](https://stackoverflow.com/questions/45530741/manage-python-multiprocessing-with-mongodb), it appears that we can create the MongoClient instance with connect=false, and then fork and then use the client to connect.

wtgee, 2017-12-20:
> According to stackoverflow, it appears that we can create the MongoClient instance with connect=false, and then fork and then use the client to connect.

We already default to `connect=False` for exactly this reason, so moving toward a single instance would be good. I haven't looked at #228 yet so will go do that.

PR #228 (2017-12-21): Share a single MongoClient instance
Caches clients using a weakref so that the client is discarded if there are no referrers.

Fixes #193.
closed 2017-12-21 by jamessynge