panoptes/POCS #230: test_observatory.py:test_status

jamessynge, 2017-12-21:
I've seen this failure several times recently.

```
____________________________________________ test_status ____________________________________________

observatory = <pocs.observatory.Observatory object at 0x7f1c539ed208>

    def test_status(observatory):
        os.environ['POCSTIME'] = '2016-08-13 10:00:00'
        status = observatory.status()
        assert 'mount' not in status
        assert 'observation' not in status
>       assert 'observer' in status
E       AssertionError: assert 'observer' in {}

pocs/tests/test_observatory.py:112: AssertionError
```

wtgee, 2017-12-21:
Odd, I've never seen that one. I'm assuming you haven't been able to identify when you see that.

wtgee, 2017-12-21:
I see this in the failing tests for #231: Travis report [here](https://travis-ci.org/panoptes/POCS/builds/319516219#L4863)

The travis logs show a warning a little further down:
```
logger.py                   37 INFO     	 Observatory initialized
----------------------------- Captured stderr call -----------------------------
2017-12-21 22:36:14 travis-job-9c760981-d745-4f56-afa3-0db3b100ce07 panoptes[22523] WARNING Can't get observatory status: 'str' object is not callable
logger.py                   40 WARNING  Can't get observatory status: 'str' object is not callable
------------------------------ Captured log call -------------------------------
logger.py                   40 WARNING  Can't get observatory status: 'str' object is not callable
```

Perhaps related to #240 I just opened?

jamessynge, 2017-12-23:
Not related to #240... I think. Instead, I've narrowed it down to two issues.

1. The tests aren't hermetic; in particular, the presence of a dome when running test_status depends upon which tests were run previously. Clearly this isn't the fault of test_status, but rather whether test_dome_simulator.py or test_astrohaven_dome.py was run previously. But, this means the config isolation is failing.
2. When the generic dome simulator is used, the simulator.py:Dome.status property throws a TypeError exception that I don't understand the source of. It says that a `str` object isn't callable, but I don't understand where it is trying to call a `str`.

jamessynge, 2017-12-24:
Filed #246 for non-hermetic issue. Fixed second issue, exception due to Dome.status.

closed 2017-12-24 by jamessynge