panoptes/POCS #235: Link focusing images to latest.js

wtgee, 2017-12-21:
Would be nice to see the focusing as it changes. Would only happen for primary camera without some work.

wtgee, 2018-02-13:
@jermainegug this might be an easy one to pick off in some of your sleepless nights :)

jermainegug, 2018-02-13:
@wtgee sounds good :)

wtgee, 2018-02-13:
Actually it's not entirely obvious what we want to see here. @AnthonyHorton, any input?

AnthonyHorton, 2018-02-14:
Hmmm, most useful would probably be the `*_final.fits` image that gets taken once the autofocus thinks it's finished.  Would be nice to be able to confirm at a glance that a decent focus was obtained.

The other images get superceded so quickly (for broadband filters, at least) they'd be gone before PAWS updated.

wtgee, 2018-02-14:
But which of the cameras would we want to view this for? Ideally we just modify it to get links for all of them I suppose, but that is more work.

AnthonyHorton, 2018-02-14:
Yes, ideally you'd check all the cameras, but that would be more work. Question is whether it would be worth it? Depends on how often there are going to be issues with focusing. Working on that...

wtgee, 2018-08-27:
This was completed at some point.

closed 2018-08-27 by wtgee