panoptes/POCS #242: test_run_wait_until_safe hangs sometimes

jamessynge, 2017-12-22:
I've taken to using a time out of **5 minutes** when running tests, and this test not infrequently hits the timeout (checked on the `develop` branch with no changes in place). Given the structure of the test, it seems like it is just hung.

A number of `test_pocs.py` tests have a similar structure, and would benefit from a redesign. Assigning to @wtgee for input on the issue and for redesign ideas, not necessarily for implementation.

wtgee, 2017-12-22:
This is a timeout on individual tests?

jamessynge, 2017-12-22:
Yes, a timeout on a single test.

jamessynge, 2017-12-22:
Details from pytest:

```
_____________________________________ test_run_wait_until_safe ______________________________________

db = <pocs.utils.database.PanMongo object at 0x7fb8341879e8>
observatory = <pocs.observatory.Observatory object at 0x7fb8340e0fd0>

    def test_run_wait_until_safe(db, observatory):
        os.environ['POCSTIME'] = '2016-08-13 23:00:00'
    
        def start_pocs():
            observatory.config['simulator'] = ['camera', 'mount', 'night']
    
            pocs = POCS(observatory,
                        messaging=True, safe_delay=15)
            pocs.db.current.remove({})
            pocs.initialize()
            pocs.logger.info('Starting observatory run')
            assert pocs.is_weather_safe() is False
            pocs.send_message('RUNNING')
            pocs.run(run_once=True, exit_when_done=True)
            assert pocs.is_weather_safe() is True
    
        pub = PanMessaging.create_publisher(6500)
        sub = PanMessaging.create_subscriber(6511)
    
        pocs_process = Process(target=start_pocs)
        pocs_process.start()
    
        # Wait for the running message
        while True:
>           msg_type, msg_obj = sub.receive_message()

pocs/tests/test_pocs.py:205: 
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
pocs/utils/messaging.py:142: in receive_message
    message = self.socket.recv_string(flags=flags)
../../../anaconda3/envs/panoptes-env/lib/python3.6/site-packages/zmq/sugar/socket.py:457: in recv_string
    msg = self.recv(flags=flags)
zmq/backend/cython/socket.pyx:693: in zmq.backend.cython.socket.Socket.recv
    ???
zmq/backend/cython/socket.pyx:727: in zmq.backend.cython.socket.Socket.recv
    ???
zmq/backend/cython/socket.pyx:145: in zmq.backend.cython.socket._recv_copy
    ???
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _

>   ???
E   Failed: Timeout >300.0s

zmq/backend/cython/checkrc.pxd:12: Failed
--------------------------------------- Captured stderr setup ---------------------------------------
2017-12-20 22:14:04 galago panoptes[27020] INFO Initializing observatory
2017-12-20 22:14:04 galago panoptes[27020] INFO 	Setting up location
2017-12-20 22:14:04 galago panoptes[27020] INFO 	Setting up mount
2017-12-20 22:14:04 galago panoptes[27020] INFO 		Using simulator mount
2017-12-20 22:14:04 galago panoptes[27020] INFO 	Setting up cameras
2017-12-20 22:14:04 galago panoptes[27020] INFO 			 Simulated Focuser (SF0577) on /dev/ttyFAKE initialised
2017-12-20 22:14:04 galago panoptes[27020] INFO 			 Simulated Focuser (SF2162) on /dev/ttyFAKE initialised
2017-12-20 22:14:04 galago panoptes[27020] INFO 	Setting up scheduler
2017-12-20 22:14:04 galago panoptes[27020] INFO 	 Observatory initialized
--------------------------------------- Captured stderr call ----------------------------------------
2017-12-20 22:14:04 galago panoptes[27171] INFO Initializing PANOPTES unit - Generic PANOPTES Unit - Mauna Loa Observatory
2017-12-20 22:14:04 galago panoptes[27171] INFO Loading state table: simple_state_table
2017-12-20 22:14:06 galago root[27171] INFO PANCHAT Hi there!
2017-12-20 22:14:06 galago root[27171] INFO PANCHAT Initializing the system! Woohoo!
2017-12-20 22:14:07 galago panoptes[27171] INFO Starting observatory run
2017-12-20 22:14:07 galago panoptes[27171] WARNING No record found in Mongo DB
2017-12-20 22:14:07 galago panoptes[27171] WARNING No record found in Mongo DB
2017-12-20 22:14:07 galago panoptes[27171] WARNING Unsafe conditions: {'is_dark': True, 'good_weather': False, 'free_space': True}
2017-12-20 22:14:07 galago panoptes[27171] WARNING No record found in Mongo DB
2017-12-20 22:14:23 galago panoptes[27171] WARNING No record found in Mongo DB
2017-12-20 22:14:39 galago panoptes[27171] WARNING No record found in Mongo DB
2017-12-20 22:14:56 galago panoptes[27171] WARNING No record found in Mongo DB
2017-12-20 22:15:12 galago panoptes[27171] WARNING No record found in Mongo DB
2017-12-20 22:15:28 galago panoptes[27171] WARNING No record found in Mongo DB
2017-12-20 22:15:44 galago panoptes[27171] WARNING No record found in Mongo DB
2017-12-20 22:16:01 galago panoptes[27171] WARNING No record found in Mongo DB
2017-12-20 22:16:17 galago panoptes[27171] WARNING No record found in Mongo DB
2017-12-20 22:16:33 galago panoptes[27171] WARNING No record found in Mongo DB
2017-12-20 22:16:49 galago panoptes[27171] WARNING No record found in Mongo DB
2017-12-20 22:17:05 galago panoptes[27171] WARNING No record found in Mongo DB
2017-12-20 22:17:22 galago panoptes[27171] WARNING No record found in Mongo DB
2017-12-20 22:17:38 galago panoptes[27171] WARNING No record found in Mongo DB
2017-12-20 22:17:54 galago panoptes[27171] WARNING No record found in Mongo DB
2017-12-20 22:18:11 galago panoptes[27171] WARNING No record found in Mongo DB
2017-12-20 22:18:27 galago panoptes[27171] WARNING No record found in Mongo DB
2017-12-20 22:18:43 galago panoptes[27171] WARNING No record found in Mongo DB
2017-12-20 22:18:59 galago panoptes[27171] WARNING No record found in Mongo DB
```

jamessynge, 2017-12-22:
I used this command to install a timeout plugin:

`pip install pytest-timeout`

wtgee, 2018-02-13:
Recent work has been done on this both in terms of changing the test as well as changing the messaging system for tests in general. Closing until more specific need identified.

closed 2018-02-13 by wtgee