panoptes/POCS #245: Wait for iOptron mount to acquire GPS signal

jamessynge, 2017-12-23:
According Kevin @ iOptron, if we **do not** send the "set lat-long" commands to an iOptron mount, it will discover its location on its own. The implication of our conversation was that if we *do* set the location explicitly, we can never find out what the GPS module in the mount has discovered.

First step: Validate what I heard from Kevin by toggling the mount off-and-on using the relays, then see if we can read the location from the mount, and how long it takes to acquire the location (which will presumably vary by location).

wtgee, 2017-12-24:
I'm curious about what this reveals. Previously we didn't have the ability to power-cycle so this wasn't really an option. Would be great to get exact GPS.

wtgee, 2020-02-16:
See also #953