panoptes/POCS #259: test_local_config affected by local configuration

jamessynge, 2017-12-28:
The title might sound obvious, but it means that one branch of the code is executed if there is no pocs_local.yaml file, which is the case if the test is not executing on a properly configured unit, and the other branch only applies to configured units. It seems like this should have a test resources directory that can contain the two yaml files.

Output from a failed test after I created a pocs_local.yaml file for testing whether pocs_shell works:

```python
    def test_local_config():
    
        _local_config_file = '{}/conf_files/pocs_local.yaml'.format(os.getenv('POCS'))
    
        if not os.path.exists(_local_config_file):
            conf = load_config(ignore_local=True)
            assert conf['name'] == 'Generic PANOPTES Unit'
    
            local_yaml = {
                'name': 'ConfTestName'
            }
            with open(_local_config_file, 'w') as f:
                f.write(yaml.dump(local_yaml))
            conf = load_config()
            assert conf['name'] != 'Generic PANOPTES Unit'
            os.remove(_local_config_file)
        else:
            conf = load_config()
>           assert conf['name'] != 'Generic PANOPTES Unit'
E           AssertionError: assert 'Generic PANOPTES Unit' != 'Generic PANOPTES Unit'
```

wtgee, 2017-12-29:
Do you have the `name` key in your existing `pocs_local.yaml`?

I see what you mean about a test resources directory however.

Another thing we might want to do is remove certain keys (`name`, `location`) from the base `pocs.yaml` such that a configured unit will be required to edit them.

jamessynge, 2017-12-29:
I think I copied pocs.yaml to pocs_local.yaml, and added the dome section,
so the name is the default.

wtgee, 2019-03-29:
Closing this as stale until more information provided.

closed 2019-03-29 by wtgee