panoptes/POCS #261: pytest failures on PAN006

jamessynge, 2017-12-29:
As mentioned in #260, I've updated PAN006 to latest commit on POCS upstream/develop (3cf2831b0dda9ecf3b8d70ab64069789d06a4c94). `pytest` shows the failures below, which I don't immediately recognize:

```python
=============================================================== FAILURES ================================================================
____________________________________________________________ test_error_exit ____________________________________________________________

config = {'cameras': {'auto_detect': True, 'devices': [{'model': 'canon_gphoto2'}, {'model': 'canon_gphoto2'}], 'primary': '14d...: <Quantity 3400.0 m>, 'gmt_offset': -600, 'horizon': <Quantity 30.0 deg>, 'latitude': <Quantity 19.54 deg>, ...}, ...}

    def test_error_exit(config):
        with pytest.raises(SystemExit):
>           Observatory(ignore_local_config=True, simulator=['none'])
E           Failed: DID NOT RAISE <class 'SystemExit'>

pocs/tests/test_observatory.py:40: Failed
____________________________________________________________ test_bad_camera ____________________________________________________________

config = {'cameras': {'auto_detect': True, 'devices': [{'model': 'canon_gphoto2'}, {'model': 'canon_gphoto2'}], 'primary': '14d...: <Quantity 3400.0 m>, 'gmt_offset': -600, 'horizon': <Quantity 30.0 deg>, 'latitude': <Quantity 19.54 deg>, ...}, ...}

    def test_bad_camera(config):
        conf = config.copy()
        simulator = hardware.get_all_names(without=['camera'])
        with pytest.raises(error.PanError):
>           Observatory(simulator=simulator, config=conf, auto_detect=True, ignore_local_config=True)
E           Failed: DID NOT RAISE <class 'pocs.utils.error.PanError'>

pocs/tests/test_observatory.py:79: Failed
_________________________________________________________ test_camera_not_found _________________________________________________________

config = {'cameras': {'auto_detect': True, 'devices': [{'model': 'canon_gphoto2'}, {'model': 'canon_gphoto2'}], 'primary': '14d...: <Quantity 3400.0 m>, 'gmt_offset': -600, 'horizon': <Quantity 30.0 deg>, 'latitude': <Quantity 19.54 deg>, ...}, ...}

    def test_camera_not_found(config):
        conf = config.copy()
        simulator = hardware.get_all_names(without=['camera'])
        with pytest.raises(error.PanError):
>           Observatory(simulator=simulator, config=conf, ignore_local_config=True)
E           Failed: DID NOT RAISE <class 'pocs.utils.error.PanError'>

pocs/tests/test_observatory.py:86: Failed
========================================== 3 failed, 225 passed, 60 skipped in 646.41 seconds ===========================================
panoptes@Panmass:/var/panoptes/POCS$ 
```

wtgee, 2017-12-29:
This looks like missing install files. Did you end up creating an anaconda environment that is not loaded?

jamessynge, 2017-12-29:
Yes I created the anaconda panoptes env, and it is activated.

wtgee, 2018-01-15:
Closing this, feel free to reopen but I think PAN006 is running now.

closed 2018-01-15 by wtgee