panoptes/POCS #264: pocs_shell emits lots of errors on startup

jamessynge, 2017-12-29:
I've updated POCS, installed dependencies for POCS & PIAA, etc. I now get lots of errors from starting pocs_shell (no local config exists, because I renamed those yaml files).

```python
panoptes@Panmass:/var/panoptes/POCS$ bin/pocs_shell 
--- Logging error ---
Traceback (most recent call last):
  File "/home/panoptes/anaconda3/envs/panoptes-env/lib/python3.6/logging/__init__.py", line 992, in emit
    msg = self.format(record)
  File "/home/panoptes/anaconda3/envs/panoptes-env/lib/python3.6/logging/__init__.py", line 838, in format
    return fmt.format(record)
  File "/home/panoptes/anaconda3/envs/panoptes-env/lib/python3.6/logging/__init__.py", line 578, in format
    s = self.formatMessage(record)
  File "/home/panoptes/anaconda3/envs/panoptes-env/lib/python3.6/logging/__init__.py", line 547, in formatMessage
    return self._style.format(record)
  File "/home/panoptes/anaconda3/envs/panoptes-env/lib/python3.6/logging/__init__.py", line 399, in format
    return self._fmt.format(**record.__dict__)
KeyError: 'fileline'
Call stack:
  File "bin/pocs_shell", line 29, in <module>
    from piaa.utils import polar_alignment
  File "<frozen importlib._bootstrap>", line 961, in _find_and_load
  File "<frozen importlib._bootstrap>", line 950, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 655, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 678, in exec_module
  File "<frozen importlib._bootstrap>", line 205, in _call_with_frames_removed
  File "/var/panoptes/PIAA/piaa/utils/__init__.py", line 11, in <module>
    from shapely.geometry import Polygon
  File "<frozen importlib._bootstrap>", line 961, in _find_and_load
  File "<frozen importlib._bootstrap>", line 950, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 655, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 678, in exec_module
  File "<frozen importlib._bootstrap>", line 205, in _call_with_frames_removed
  File "/home/panoptes/anaconda3/envs/panoptes-env/lib/python3.6/site-packages/shapely/geometry/__init__.py", line 4, in <module>
    from .base import CAP_STYLE, JOIN_STYLE
  File "<frozen importlib._bootstrap>", line 961, in _find_and_load
  File "<frozen importlib._bootstrap>", line 950, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 655, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 678, in exec_module
  File "<frozen importlib._bootstrap>", line 205, in _call_with_frames_removed
  File "/home/panoptes/anaconda3/envs/panoptes-env/lib/python3.6/site-packages/shapely/geometry/base.py", line 17, in <module>
    from shapely.coords import CoordinateSequence
  File "<frozen importlib._bootstrap>", line 961, in _find_and_load
  File "<frozen importlib._bootstrap>", line 950, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 655, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 678, in exec_module
  File "<frozen importlib._bootstrap>", line 205, in _call_with_frames_removed
  File "/home/panoptes/anaconda3/envs/panoptes-env/lib/python3.6/site-packages/shapely/coords.py", line 8, in <module>
    from shapely.geos import lgeos
  File "<frozen importlib._bootstrap>", line 961, in _find_and_load
  File "<frozen importlib._bootstrap>", line 950, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 655, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 678, in exec_module
  File "<frozen importlib._bootstrap>", line 205, in _call_with_frames_removed
  File "/home/panoptes/anaconda3/envs/panoptes-env/lib/python3.6/site-packages/shapely/geos.py", line 67, in <module>
    LOG.debug("Found GEOS DLL: %r, using it.", _lgeos)
Message: 'Found GEOS DLL: %r, using it.'
Arguments: (<CDLL '/home/panoptes/anaconda3/envs/panoptes-env/lib/python3.6/site-packages/shapely/.libs/libgeos_c-bd8d3f16.so.1.10.2', handle 26ee5e0 at 0x7fb28054af98>,)
--- Logging error ---
Traceback (most recent call last):
  File "/home/panoptes/anaconda3/envs/panoptes-env/lib/python3.6/logging/__init__.py", line 992, in emit
    msg = self.format(record)
  File "/home/panoptes/anaconda3/envs/panoptes-env/lib/python3.6/logging/__init__.py", line 838, in format
    return fmt.format(record)
  File "/home/panoptes/anaconda3/envs/panoptes-env/lib/python3.6/logging/__init__.py", line 578, in format
    s = self.formatMessage(record)
  File "/home/panoptes/anaconda3/envs/panoptes-env/lib/python3.6/logging/__init__.py", line 547, in formatMessage
    return self._style.format(record)
  File "/home/panoptes/anaconda3/envs/panoptes-env/lib/python3.6/logging/__init__.py", line 399, in format
    return self._fmt.format(**record.__dict__)
KeyError: 'fileline'
[... 130 lines cut ...]
  File "<frozen importlib._bootstrap_external>", line 678, in exec_module
  File "<frozen importlib._bootstrap>", line 205, in _call_with_frames_removed
  File "/home/panoptes/anaconda3/envs/panoptes-env/lib/python3.6/site-packages/shapely/geometry/base.py", line 17, in <module>
    from shapely.coords import CoordinateSequence
  File "<frozen importlib._bootstrap>", line 961, in _find_and_load
  File "<frozen importlib._bootstrap>", line 950, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 655, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 678, in exec_module
  File "<frozen importlib._bootstrap>", line 205, in _call_with_frames_removed
  File "/home/panoptes/anaconda3/envs/panoptes-env/lib/python3.6/site-packages/shapely/coords.py", line 8, in <module>
    from shapely.geos import lgeos
  File "<frozen importlib._bootstrap>", line 961, in _find_and_load
  File "<frozen importlib._bootstrap>", line 950, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 655, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 678, in exec_module
  File "<frozen importlib._bootstrap>", line 205, in _call_with_frames_removed
  File "/home/panoptes/anaconda3/envs/panoptes-env/lib/python3.6/site-packages/shapely/geos.py", line 76, in <module>
    free = load_dll('c').free
  File "/home/panoptes/anaconda3/envs/panoptes-env/lib/python3.6/site-packages/shapely/geos.py", line 50, in load_dll
    LOG.debug("DLL: %r", dll)
Message: 'DLL: %r'
Arguments: (<CDLL 'libc.so.6', handle 7fb29ac919c8 at 0x7fb27fd75208>,)
Welcome to POCS Shell! Type ? for help
POCS > 
```

wtgee, 2017-12-29:
The line `KeyError: 'fileline'` is from the new logging change. The `log.yaml` specifies that but that means the `logger.py` file is not finding it correctly.

jamessynge, 2017-12-30:
Fixed by #265.

closed 2017-12-30 by jamessynge