panoptes/POCS #275: Arduino counter overflow

jamessynge, 2017-12-30:
The "num" field is apparently an int (16-bits on the Arduino Micro), so is overflowing fairly often.

{"name":"camera_board", "count":444741711, "num":-25746, "inputs":5, "camera_00":1, "camera_01":1, "accelerometer":{"x":-7.01, "y":6.90, "z":1.63, "o": 6}, "humidity":37.40, "temp_00":-10.20}

wtgee, 2017-12-30:
We don't have a `num` field that we report. Is that something you added in your version?

jamessynge, 2018-01-01:
Yes. There was a count field that shows the output from millis(), and I added num to show the count of reports that have been generated. Ideally (IMHO) we'd rename these to millis and report_num.

wtgee, 2018-02-13:
I believe this has been correctly handled, closing for now. Reopen if needed.

closed 2018-02-13 by wtgee