panoptes/POCS #281: Enable Arduino IDE to work with camera & telemetry boards

jamessynge, 2017-12-31:
We have some of the required libraries copied into the POCS repo (but not all), however the Arduino IDE doesn't make it easy to use libraries that aren't installed in a user or system-wide Arduino directory.

Fix this by using symlinks. This works on Linux, should work on MacOS, and may even work on modern versions of Windows.

wtgee, 2018-01-01:
Note that these were removed as part of consolidation process. Just made notes yesterday regarding this specific item: https://github.com/panoptes/POCS/issues/107

I'm a little bit sceptical about having these libraries in our repo but it does facilitate building. We could also think about just having a script that installs them, which would keep them out of our repo but make them available to user.

Note that lately I've been using the web-based Arduino IDE, which will automatically handle the libraries. However I realize that's not ideal for an actual unit.

jamessynge, 2018-01-01:
I understand the skepticism. However, since these libraries are stable for our purposes, are very small, and are not easy to automatically add to the Arduino install (i.e. its tools are primarily aimed at the GUI, not CLI).

PR #282 (2018-01-01): Use symlinks to get around Arduino IDE limitations
Add dallas-temperature-control library, used by the telemetry board sketch.

Fixes #281.
closed 2018-01-01 by jamessynge