panoptes/POCS #284: Improve Arduino enumeration

jamessynge, 2017-12-31:
In the review of PR #280, Wilfred suggested using PySerial to enumerate the set of ports:

> We can also grab connected ports and just check those: https://pythonhosted.org/pyserial/tools.html?highlight=comport#serial.tools.list_ports.comports. We could then check vendor id and product id. The arduino uno even reports a serial_number correctly although I don't think the micros do.
>
> That could probably help clean up some of the _auto_detect_port below.

For testing, I'd like to enable the use of serial handlers, which are probably not supported by list_ports.

wtgee, 2018-01-01:
> For testing, I'd like to enable the use of serial handlers, which are probably not supported by list_ports.

Not sure I follow on this comment. The auto detect feature wouldn't work with the tests anyway, right?

jamessynge, 2018-01-15:
Fixed with #320.

closed 2018-01-15 by jamessynge