panoptes/POCS #304: Add tests for pocs_shell

jamessynge, 2018-01-06:
pocs_shell is quite large, but there are no tests of it.

jamessynge, 2019-07-21:
Assigning to @ImmortalRabbit following our discussion of his decoupling changes which remove Dome and Scheduler creation from Observatory. There have not been corresponding changes (yet) to pocs_shell, so it seems likely that the current code might not work properly with POCS/bin/pocs_shell.

I propose that we extract the code that isn't related to command line parsing from pocs_shell, and move it to a top-level "driver" class, along with some of the attributes of PocsShell (e.g. pocs, msg_subscriber, etc.). For example, do_setup_pocs would parse the command line, then call a setup_pocs method of the driver class. This class will be testable, and might actually allow us to test the command line parsing of pocs_shell by creating a fake driver and injecting it into pocs_shell. Altogether, this means refactor PocsShell, extracting portions of it out into a helper called PocsDriver.

- [ ] Create class PocsDriver in POCS/pocs/pocs_driver.py, move the msg_* and cmd_* attributes to the driver, along with the methods that most directly interact with them (e.g. start_messaging, etc.).
- [ ] Write pytests of the new messaging related methods in PocsDriver.
- [ ] Add a pocs_driver attribute to PocsShell, initialized to an instance of PocsDriver.
- [ ] Remove the msg_* and cmd_* attributes of pocs_shell, and modify pocs_shell to call the methods in PocsDriver instead.
- [ ] Move the helper functions at the bottom of pocs_shell (e.g. get_observation) to the driver file (necessary for those helper functions that are used by the new methods of PocsDriver).
- [ ] Write pytests for the helper methods.
- [ ] Move the attribute pocs from PocsShell to PocsDriver, and create methods for the non-CLI parsing activities that PocsShell was doing with pocs.
- [ ] Write pytests of the new pocs related methods in PocsDriver.
- [ ] Remove the pocs attribute of pocs_shell, and modify pocs_shell to call the appropriate methods in PocsDriver instead.

Bonus items:

- [ ] Create an AbstractPocsDriver with abstract methods that are needed by PocsShell, use as the base class of PocsDriver.
- [ ] Create a MockPocsDriver (subclass of AbstractPocsDriver) (@wtgee can provide pointers regarding mocking).
- [ ] Allow injection of the MockPocsDriver in PocsShell
- [ ] Write tests of PocsShell where we pass in a command line string to the parsing code, and confirm that the MockPocsDriver was called with the appropriate args.

Note also that pocs_shell contains DriftShell, which could also similar treatment.

wtgee, 2022-01-15:
Closing as stale.

closed 2022-01-15 by wtgee