panoptes/POCS #311: Compilation warning from telemetry_board.ino

jamessynge, 2018-01-10:
I only see this warning on PAN006, not my laptop or travis:

```
In file included from /var/panoptes/POCS/resources/arduino_files/telemetry_board/telemetry_board.ino:28:0:
sketch/dallas_temperature_handler.h: In instantiation of 'void DallasTemperatureHandler<kMaxSensors>::PrintDeviceInfo() [with unsigned int kMaxSensors = 3u]':
/var/panoptes/POCS/resources/arduino_files/telemetry_board/telemetry_board.ino:391:32:   required from here
sketch/dallas_temperature_handler.h:77:62: warning: invalid conversion from 'const uint8_t* {aka const unsigned char*}' to 'uint8_t* {aka unsigned char*}' [-fpermissive]
         uint8_t resolution = dt_.getResolution(device.address);
                                                              ^
In file included from sketch/dallas_temperature_handler.h:5:0,
                 from /var/panoptes/POCS/resources/arduino_files/telemetry_board/telemetry_board.ino:28:
sketch/DallasTemperature.h:104:11: note: initializing argument 1 of 'uint8_t DallasTemperature::getResolution(uint8_t*)'
   uint8_t getResolution(uint8_t*);
           ^
```

wtgee, 2018-01-10:
Can you include the command you used to compile it? I'm going to try similar on PAN001

jamessynge, 2018-01-10:
I can check later, but I think it was:
```
~/arduino-1.8.5/arduino --upload --port /dev/ttyACM1 --board arduino:avr:micro telemetry_board.ino
```
The port needs to be changed based on which port your telemetry board is connected to. The command was executed from `$POCS/resources/arduino_files/telemetry_board`.

wtgee, 2018-01-11:
Note that I see this on PAN001 as well.

wtgee, 2018-01-11:
Confirmed I'm **not** seeing it on my local.

jamessynge, 2018-01-12:
The latest version of the DallasTemperature code declares the input device array as const, which must be what I was using when I wrote that code last summer, on a different laptop. I'll update and retest.

PR #338 (2018-01-14): Update DallasTemperature and OneWire libraries.
Fetched latest from https://github.com/milesburton/Arduino-Temperature-Control-Library
The github repo now includes OneWire, so storing both libraries in the
same directory.

Closes #311.
closed 2018-01-14 by jamessynge