panoptes/POCS #316: Temperatures not reading on PAN001

wtgee, 2018-01-11:
The DS18B20s are not reading on the telemetry board after updating to most recent code. Perhaps related to #311

jamessynge, 2018-01-11:
I'll make up a board with a couple of the temp sensors so that I can validate the code.

wtgee, 2018-01-11:
I am going to try and adapt the trucker board to the shared libraries in which case I have a board already built up to test. Will see if I can do this weekend.

wtgee, 2018-01-12:
Definitely part of #311. The compilation warnings relate to a section of the code that is printing the resolution for the DHT as part of `PrintDeviceInfo`. I simply commented out that line and compilation worked without complaints. Uploading to telemetry board then gets me temperature (and current) readings.

Closing as non-issue in favor of fixing #311

closed 2018-01-12 by wtgee