panoptes/POCS #318: Prevent Arduinos from locking up

jamessynge, 2018-01-11:
I screwed something up with the telemetry board on PAN006 a while agosuch that it wouldn't respond.
This should be fixed in current code, but it would be a good idea to keep ourselves from screwing up. One solution is a watchdog timer. See: https://tushev.org/articles/arduino/5/arduino-and-watchdog-timer

According to that page, this code would trigger a reset if loop() doesn't run every 8 seconds:
```c++
#include <avr/wdt.h>
void setup(){
wdt_enable(WDTO_8S);
}
void loop(){
//do stuff here
wdt_reset();
}
```

wtgee, 2018-01-11:
I played with the watchdog timer in the beginning (three years ago) and seem to remember I couldn't get it to work on the micro. Worth exploring again though.

kmeagle1515, 2019-04-18:
@jamessynge @wtgee for timer specific threshold values are available. That mainly depends upon our requirement. So what threshold should be ideal for our system?

PR #1035 (2020-11-14): Arduino board cleanups
We have traditionally used a series of different Arduino sketches to handle the various electronics boards that we have. Each of these was customized to the specific board, one for each iteration of the hardware.  These have had many different names and has generally been very confusing.

Going forward we will be using the [Firmata framework](http://firmata.org/wiki/Main_Page) to abstract the functionality of each board, moving the logic off the arduino and onto Python.  

We will be implementing this with a slight variation of the standard Firmata sketch via [Firmata Ex [... cut]
closed 2020-11-14 by wtgee