panoptes/POCS #321: Cleanup log files

wtgee, 2018-01-12:
Because we write new log files for every process there can very quickly be a lot of log files in the `$PANLOG` dir. Of particular note is the `plot_weather.log`, which generates a new log file every 5 minutes of the day.

jamessynge, 2018-01-12:
Would it also help if we created sub-directories for each program, and placed the symlink in the parent $PANLOG directory?

wtgee, 2018-01-12:
For the weather I'm mostly just concerned about the shear number of them, which would be overwhelming. I was thinking we could maybe pass an option that would allow for appending to existing files in cases such as this.

wtgee, 2020-06-01:
Log files have changed with #951. Please open new issues for any log related items.

closed 2020-06-01 by wtgee