panoptes/POCS #326: Define consistent JSON format for boards.

jamessynge, 2018-01-12:
We need to have the two boards use the same basic structure for their output. They should have a version identifier, relay values should be identified as such.

wtgee, 2018-01-12:
Just a note that these are mostly passed through, which means it relates somewhat to this conversation: https://github.com/orgs/panoptes/teams/pocs-developers/discussions/6

jamessynge, 2018-09-22:
Resurrecting this issue...

Let's start with some facts, based on the currently committed sketches and the January 2017 electronics:

- The camera board is set to emit a report [once every second](https://github.com/panoptes/POCS/blob/develop/resources/arduino_files/camera_board/camera_board.ino#L148) (specifically, every 1000 ms).
- The telemetry board is set to emit a report [once every two seconds](https://github.com/panoptes/POCS/blob/develop/resources/arduino_files/telemetry_board/telemetry_board.ino#L52).
- Each report is a single line of text terminated with a CR LF (newline, linefeed).
- The reports are in JSON object format; the only white space is the space character between property:value pairs, and the terminating CR LF.
- The first property in each report is "name", currently with one of these two values "telemetry_board" or "camera_board."
- Reports are written to the local db mostly unmodified, but with some extra info added.
- A camera board report has these properties:
  - name: "camera_board"
  - count: The number of milliseconds since the `setup()` method finished. _This is problematic because it the code doesn't deal with wrap around._ 
  - num: The report number; starts at one, increments with each report.
  - inputs_received: Number of times that `Serial.available() > 0` was true. This is for debugging the parsing of inputs (i.e. relay changes).
  - camera_00: Whether the relay for camera 0 is open/off (value=0) or closed/on (value=1).
  - camera_01: Whether the relay for camera 1 is open/off (value=0) or closed/on (value=1). 
  - accelerometer: The value is a JSON object with the readings from the [MMA8451](https://www.adafruit.com/product/2019) 3-axis accelerometer; there are four properties, x, y, z and o (for orientation).
  - humidity: The relative humidity as determined by the [DHT22](https://www.adafruit.com/product/385) or [AM2302](https://www.adafruit.com/product/393) temperature and humidity sensor (the latter contains the former). Note that relative humidity varies by temperature, with warmer air able to hold more moisture without becoming saturated (i.e. without it starting to condense on surfaces or to rain).
  - temp_00: The temperature (Celcius) as determined by the DHT22 or AM2302 sensor. The numbering is to allow for the extra temperature sensors reported by the telemetry board.

- The telemetry board reports have these properties:
  - name: "telemetry_board"
  - millis: The number of milliseconds since the sketch started. This will reset to zero after 49.71 days: the 32-bit unsigned value can only count that high, then wraps around to zero and continues counting. It is common for the software in an embedded microcontroller, such as in the Arduino, to run continuously through many such intervals.
  - report_num: The report number; starts at one, increments with each report. This is unlikely to wrap around as it would take 272 years!
  - ver: A string indicating the version of the Arduino sketch. Currently "2018-01-14". _It might be good to include the git commit id in some way._
  - power: This is a JSON object with properties related to the state of relays/power lines. In each case, the value indicates if the relay/power line is open/off (value=0) or closed/on (value=1).
    - computer: State of the relay controlling power to the NUC.
    - fan: State of the relay controlling power to the cooling/ventilation fan.
    - mount: State of the relay controlling power to the equatorial mount.
    - cameras: State of the relay controlling power to the camera box.
    - weather: State of the relay controlling power to the weather sensors.
    - main: Does the AC power line have power? Used for detecting power failures.
  - current: This is a JSON object with properties recording the value from the Analog to Digital converter applied to the voltage output of the electrical current sensors. These are in the range 0 through 1023, and their real world meaning is based on the offset and scale settings of sensors. The JSON object has these properties:
    - main: DC input current from the power supply to the V0 power board. May be over 5A, but unlikely.
    - fan: DC output current to the fan (should be well under 500mA).
    - mount: DC output current to the equatorial mount (should be under 1A most of the time).
    - cameras: DC output current to the camera box (i.e. to the cameras, camera_board and USB hub that are all in the camera box).
  - amps: This is a JSON object with the same set of property names as in current, but with the values scaled from ADU (analog-digital units) to amps... but only if the constants in the sketch (see `main_amps_mult`) are accurate, which is unlikely because the original current sensors were very hard to calibrate. The values in the sketch need to be modified for the local unit, but we don't provide a way to do that without modifying this sketch.
  - humidity: The relative humidity as determined by the [DHT22](https://www.adafruit.com/product/385) or [AM2302](https://www.adafruit.com/product/393) temperature and humidity sensor (the latter contains the former). Note that relative humidity varies by temperature, with warmer air able to hold more moisture without becoming saturated (i.e. without it starting to condense on surfaces or to rain).
  - temp_00: The temperature (Celcius) as determined by the DHT22 or AM2302 sensor. 
[... cut ...]
- We've had some difference of opinion regarding how complex or simple the Arduino code should be. I'm a C++ programmer, so when I went to overhaul some of the code, I added classes and abstractions, but that can be intimidating to new comers, and we want students that are comfortable with the classic Blink sketch to also have a hope of understanding our sketches.
- I wrote the more complex code because peas_shell wasn't successfully reading all of the reports. I think this was probably because the SerialData class wasn't very good at the time, but I've since addressed the problems that caused it to get confused by the input... at least mostly.

wtgee, 2018-09-22:
Thanks for putting this together. Quick comment, more later:

As mentioned in another Issue, I think we should be coming up with a standard JSON output scheme for hardware devices in general rather than anything specific to the arduinos or the sketches. In the simple case this is because, for instance, PAN008 uses a Raspberry Pi instead of an arduino in the camera box. In the bigger case, I think we ultimately want all devices (mount, camears, etc) to essentially be emitting their status output (probably via zmq) and they should all have a similar setup.

PR #1035 (2020-11-14): Arduino board cleanups
We have traditionally used a series of different Arduino sketches to handle the various electronics boards that we have. Each of these was customized to the specific board, one for each iteration of the hardware.  These have had many different names and has generally been very confusing.

Going forward we will be using the [Firmata framework](http://firmata.org/wiki/Main_Page) to abstract the functionality of each board, moving the logic off the arduino and onto Python.  

We will be implementing this with a slight variation of the standard Firmata sketch via [Firmata Ex [... cut]
closed 2020-11-14 by wtgee