panoptes/POCS #331: Move arduino verifcation above other tests in travis

wtgee, 2018-01-13:
Failing travis tests should come last in the file. Currently there is a lot of successful output from the arduino verification, which can be a bit confusing.

jamessynge, 2018-01-15:
I'll do the move because I think it is helpful, but of course we have no idea which operation will fail.

wtgee, 2018-01-15:
> I'll do the move because I think it is helpful, but of course we have no idea which operation will fail.

You mean a travis test vs an arduino test? Agree we don't know which will fail but I think we will be altering the travis tests more than arduino (in the long run). Thanks for moving.

The other option would be to wrap the verification in a subprocess and run it is a travis test but I think that's a bit overkill where this works fine.

PR #353 (2018-01-15): Verify Arduino sketches before running pytest.
Fixes #331.
closed 2018-01-15 by wtgee