panoptes/POCS #335: Cleanup camera methods

wtgee, 2018-01-14:
Currently the `take_observation`, `take_exposure`, and `process_exposre` methods in the base camera class are not implemented, leading to a lot of duplication in `canon_gphoto2` and `sbig` (and probably the coming FLI). This needs to be cleaned up so the base class is handling the basic mechanics.

AnthonyHorton, 2018-01-14:
I have been reducing some of the duplication of code between `sbig` & `canon_gphoto2`, and avoiding adding to it with `fli`, but I've been doing it by adding utility functions as methods of the base class rather than by adding base class implementations of `take_observation`, `take_exposure` and `process exposure`. Was going to open a PR for my branch soon (the FLI camera support is mostly done, but not fully tested and not as robust as it should be, yet).  Perhaps I should do it now to facilitate discussion on these sorts of issues?

wtgee, 2018-01-14:
That sounds fine too. Open up the PR when you are ready. Some things have changed in there (stupid EQUINOX problems again) so you might need to do some merging from `upstream/develop` (note we are not doing `rebase` for the most part but merging - following the [Feature Development Process](https://github.com/panoptes/POCS/wiki/Feature-Development-Process) has worked pretty well to avoid all our problem scenarios from earlier).

wtgee, 2018-02-13:
This was mostly taken care of in #363. Additional items should have their own Issue.

closed 2018-02-13 by wtgee