panoptes/POCS #339: Add a utility program for listing info about hardware

jamessynge, 2018-01-14:
List camera, Arduinos, mount, dome. Print status, port, serial numbers, etc., as appropriate.

jamessynge, 2018-01-14:
Wilfred suggested:

FWIW, and regarding comment (#320) above about separate utils for discovering devices, I have mostly been avoiding making the module callable via `if __main__` _unless_ they are in the utils folder. However, that would make development a little annoying as you would have to run a util file to test the sensors file.  Still, could end up looking something like:
```
python pocs/utils/devices.py --list_arduinos
python pocs/utils/devices.py --list_cameras
python pocs/utils/devices.py --list_attached_hardware
```
etc.

wtgee, 2020-02-16:
Could also be a part of https://github.com/panoptes/panoptes-tutorials/issues/48

PR #1035 (2020-11-14): Arduino board cleanups
We have traditionally used a series of different Arduino sketches to handle the various electronics boards that we have. Each of these was customized to the specific board, one for each iteration of the hardware.  These have had many different names and has generally been very confusing.

Going forward we will be using the [Firmata framework](http://firmata.org/wiki/Main_Page) to abstract the functionality of each board, moving the logic off the arduino and onto Python.  

We will be implementing this with a slight variation of the standard Firmata sketch via [Firmata Ex [... cut]
closed 2020-11-14 by wtgee