panoptes/POCS #341: Add utility for aligning of cameras

jamessynge, 2018-01-15:
It is clear that the PAN006 cameras aren't pointed in quite the same direction, nor are they aimed parallel to the RA axis when the mount is aimed at the pole (i.e. after polar alignment). It would be helpful to have some UI that gives a measure of how far left or right to rotate each camera.

(This and a mention by Wilfred that he wasn't sure which camera was controlled by each Arduino pin, makes me think we should have a canonical position for each camera, e.g. left for Cam00 and right for Cam01.)

wtgee, 2018-01-15:
Just a note that it's not critical that they do point exactly the same although the utility would be nice anyway. However, accessing the cameras isn't very easy so we would definitely want to do this before final setup.

A name for each position is a good idea.

wtgee, 2020-02-16:
Could also be a part of https://github.com/panoptes/panoptes-tutorials/issues/48