panoptes/POCS #343: Document how to upload to Arduinos from Linux

jamessynge, 2018-01-15:
User needs permission to use device (e.g. /dev/ttyACM0).
May need to remove modem manager: `sudo apt-get purge modemmanager`
There is advice on arduino.cc regarding using Linux for Arduino development, and we should provide pointers to that.

kmeagle1515, 2019-03-21:
hello sir!
Where will the documentation be included, in **README** or you want it in a different folder?

kmeagle1515, 2019-03-25:
I have made a pr please review it.

AnthonyHorton, 2019-03-26:
@kmeagle1515 For future reference, you should link related Issues and Pull Requests. Just use the Issue/PR number in the description or a comment and GitHub will automatically convert it to a link, e.g. the PR you're referring to here is #802.  Note, this *doesn't* work if you only do it in the title (as in your PR).

wtgee, 2019-03-26:
> @kmeagle1515 For future reference, you should link related Issues and Pull Requests. Just use the Issue/PR number in the description or a comment and GitHub will automatically convert it to a link, [...]

And we actually just added PR templates as part of #156.

kmeagle1515, 2019-04-18:
@jamessynge  @wtgee this issue has been solved #802 .

closed 2019-04-19 by wtgee