panoptes/POCS #347: PEAS shell: command to clear buffer on arduino

wtgee, 2018-01-15:
If the arduino output gets behind would be nice to be able to manually clear buffer.

jamessynge, 2018-01-15:
I think we have two alternatives, where I prefer the first:

1. Separate the reading and storing of the two Arduinos, so that they don't have to output reports at precisely the same rate. This would of course require a separate thread for each of the Arduinos.
2. Read all available bytes from each arduino, and extract the latest complete line from those bytes, allowing us to skip past any backlog.

wtgee, 2018-01-15:
Yes, I agree that first option is better. This goes along with the splitting of ArduinoSerialMonitor.

But I still think having a command to do it manually could come in handy if ever needed. Are you thinking these alternatives would eliminate that need?

kmeagle1515, 2019-04-02:
pyserial commands like reset_input_buffer( ) and  reset_output_buffer( ) may come in handy.
Maybe there is a need for a new function like clear_buffer ( ) under which these commands will be executed because peas_shell doesn't have one. 

Also i have noticed a line
https://github.com/panoptes/POCS/blob/4ef6e70e0484156c67b5c8178235b2a9987cca58/bin/peas_shell#L210
here flushInput( ) is supported for python versions 2.x the latest commands is reset_input_buffer( ) 
https://pyserial.readthedocs.io/en/latest/pyserial_api.html#serial.Serial.reset_input_buffer

wtgee, 2022-01-15:
Stale.

closed 2022-01-15 by wtgee