panoptes/POCS #348: Fix "Waiting for images" timer

wtgee, 2018-01-15:
It outputs debug every 15 seconds but it's not in sync very well. For a 120 second image the output usually stops at about 90 sec.

wtgee, 2018-01-15:
Output from session:

```
D0115 08:22:33.910 observatory.py:235   Exposing for camera: Cam00
D0115 08:22:33.910 canon_gphoto2.py:121 image_id: PAN001_14d3bd_20180115T082233
D0115 08:22:33.910 canon_gphoto2.py:174 Taking 120.0 second exposure on Cam00: /var/panoptes/images/fields/Wasp36/14d3bd/20180115T081653//20180115T082233.cr2
D0115 08:22:33.918 observatory.py:235   Exposing for camera: Cam01
D0115 08:22:33.920 canon_gphoto2.py:121 image_id: PAN001_ee04d1_20180115T082233
D0115 08:22:33.920 canon_gphoto2.py:174 Taking 120.0 second exposure on Cam01: /var/panoptes/images/fields/Wasp36/ee04d1/20180115T081653//20180115T082233.cr2
D0115 08:22:33.931 observing.py:25      Waiting for images: 0.0 seconds
D0115 08:22:56.364 observing.py:25      Waiting for images: 15.0 seconds
D0115 08:23:18.770 observing.py:25      Waiting for images: 30.0 seconds
D0115 08:23:41.223 observing.py:25      Waiting for images: 45.0 seconds
D0115 08:24:03.676 observing.py:25      Waiting for images: 60.0 seconds
D0115 08:24:26.098 observing.py:25      Waiting for images: 75.0 seconds
D0115 08:24:39.919 canon_gphoto2.py:218 Processing PAN001_14d3bd_20180115T082233
D0115 08:24:39.919 canon_gphoto2.py:221 Extracting pretty image
D0115 08:24:39.928 canon_gphoto2.py:226 Converting CR2 -> FITS: /var/panoptes/images/fields/Wasp36/14d3bd/20180115T081653//20180115T082233.cr2
D0115 08:24:39.933 canon_gphoto2.py:218 Processing PAN001_ee04d1_20180115T082233

```

jamessynge, 2018-01-16:
Not sure if relevant, but #357 is related to timeouts.

wtgee, 2018-01-16:
> Not sure if relevant, but #357 is related to timeouts.

We have the process with the stdout, so I think in this case we should actually just grab the output timer from gphoto2 directly so it reflects what the camera is doing.

wtgee, 2018-08-27:
Closed by #559

closed 2018-08-27 by wtgee