panoptes/POCS #369: Determine how Horizon Altitude score should work

wtgee, 2018-01-16:
We currently just return a score of 100 if it is not vetoed. Determine how best to use this.

jermainegug, 2018-01-23:
Once we can find out how to check the Horizon constraint over the whole duration of an observation we can have interchangeable variables to either veto the target, give it low score. 

Maybe something like this?
```
if obstructed_duration > (veto_time_portion * time):
    veto = True
elif obstructed_duration > (low_score_time_portion * time)
    score = short_obstruction_score # must be less that 100
else:
    score = 100
```

So then if there are no better targets we can still observe those which might only have a short, partial obstruction where we can choose the range of those targets in a config or better yet have an extra setting on PAWS to select the time portions and scores for obstructed fields.
This should only be applicable for targets that are being observed for a long amount of time or as @jamessynge said, for a field image that is to be imaged for a lengthy amount of time. Then for targets that are only being observed for a short amount of time we just want to veto if there is an obstruction.

wtgee, 2018-01-23:
@jermainegug Yes, I like that approach and think it's a good start. Let's focus on the entire time duration (#365) first as I think for right now the "veto or nothing" approach is good enough. The other bit would have to be done first anyway.