panoptes/POCS #379: Evaluate gusty windy behavior

wtgee, 2018-01-17:
It's currently very windy for PAN001 so safety check fails and the unit is parking itself. However then the wind seems to die down for at least 15 minutes so it becomes safe. Then it very quickly gets gusty again so shuts down after about 3 images. Has done this 4 times so far.

Maybe if we shut down due to bad weather more than 3 times in an hour we should wait longer to come back up?

joshwalawender, 2018-01-17:
I'd recommend just setting that latch time to be longer instead to keep the logic easier to follow.  Maybe try 30 minutes or an hour?

wtgee, 2018-01-17:
Yes, probably a good idea. Although it's now happened 4 times in a row which means even 30 would get hit with this. Seems like waiting an hour to check weather is a long time but I guess not.

wtgee, 2018-01-17:
I changed to 30 for now on the local config.

joshwalawender, 2018-01-17:
I think we find a reasonable compromise like 30.  As with all automated systems, if we set the goal that the system will behave as a human would (i.e. give up if the wind is bouncing around the threshold and does not seem to be improving) then that means we're developing human level AI. 😁 Instead, I think we should accept that there will be seemingly stupid behavior from the system sometimes and that is a natural consequence of a system like this.

wtgee, 2018-01-18:
Yes, I set to 30 and it is working fine. Going to close this issue although I might bump the default in the generic config file too.

closed 2018-01-18 by wtgee