panoptes/POCS #392: Mark horizon from current image

wtgee, 2018-01-19:
One feature that might be nice with the Horizon constraint is to somehow just push a button in PAWS when the cameras are currently viewing an obstruction and it would stop viewing that target and write out the current alt/az to the config list.

jermainegug, 2018-01-23:
Might need some sort of interactive image interface in PAWS where you select two points and it can write those coordinates as a list. But I think that might be too much effort to make opposed to just entering the coordinates in the config manually or even just making a tab in PAWS **Input New Obstruction Coordinates**, where you enter alt/az for two points and it writes that in the config list.

I don't think writing the current alt/az to the config list would be useful, since the `Horizon` class needs two points and I can't see how it would work for only one.

wtgee, 2018-01-23:
An interactive feature is definitely overkill for now. The one nice feature would be to mark the current image from PAWS (#392) but that does have the problem of only having one point.

jermainegug, 2018-01-23:
Definitely, especially once we have the code working to get Horizon obstructions from the images. 

Maybe make a new log file, say `obstructions.log` and we just have a button that will write `obstruction was found in <image>` in the log file or we could have the FITS header have `OBSTRUCTION=False` and then pressing the button would just make it `True`?

jermainegug, 2018-01-23:
Can just start this off with marking the image so that we know it has an obstruction in it. Then I guess once the Horizons code is done you could possibly just have a function that says marked images are to be parsed through `get_obstruction_from_image` (#364).

wtgee, 2020-02-17:
Closing as duplicate of #364

closed 2020-02-17 by wtgee