panoptes/POCS #395: UserWarning: Problem creating timelapse

jamessynge, 2018-01-19:
Error from PAN006, which I'm guessing indicates that some software is required that POCS doesn't have in its requirements.

```
 6510
Starting POCS - Press Ctrl-c to interrupt
/var/panoptes/POCS/pocs/utils/images/__init__.py:206: UserWarning: Problem creating timelapse: /var/panoptes/images/timelapse/Wasp35_6575fc_20180118T232417.mp4
  warn("Problem creating timelapse: {}".format(fn_out))
/var/panoptes/POCS/pocs/utils/images/__init__.py:206: UserWarning: Problem creating timelapse: /var/panoptes/images/timelapse/Wasp35_7bab97_20180118T232417.mp4
  warn("Problem creating timelapse: {}".format(fn_out))
/var/panoptes/POCS/pocs/utils/images/__init__.py:206: UserWarning: Problem creating timelapse: /var/panoptes/images/timelapse/HatP30_6575fc_20180119T020703.mp4
  warn("Problem creating timelapse: {}".format(fn_out))
/var/panoptes/POCS/pocs/utils/images/__init__.py:206: UserWarning: Problem creating timelapse: /var/panoptes/images/timelapse/HatP30_7bab97_20180119T020703.mp4
  warn("Problem creating timelapse: {}".format(fn_out))
/var/panoptes/POCS/pocs/utils/images/__init__.py:206: UserWarning: Problem creating timelapse: /var/panoptes/images/timelapse/Wasp104_6575fc_20180119T051913.mp4
  warn("Problem creating timelapse: {}".format(fn_out))
/var/panoptes/POCS/pocs/utils/images/__init__.py:206: UserWarning: Problem creating timelapse: /var/panoptes/images/timelapse/Wasp104_7bab97_20180119T051913.mp4
  warn("Problem creating timelapse: {}".format(fn_out))
/var/panoptes/POCS/pocs/utils/images/__init__.py:206: UserWarning: Problem creating timelapse: /var/panoptes/images/timelapse/Wasp104_6575fc_20180119T074529.mp4
  warn("Problem creating timelapse: {}".format(fn_out))
/var/panoptes/POCS/pocs/utils/images/__init__.py:206: UserWarning: Problem creating timelapse: /var/panoptes/images/timelapse/Wasp104_7bab97_20180119T074529.mp4
  warn("Problem creating timelapse: {}".format(fn_out))
```

wtgee, 2018-01-19:
Some of these have started showing up in travis recently too. I'll double check that PAN001 is still working ok

wtgee, 2018-01-19:
Working fine on PAN001.  Python module is `ffmpy`, underlying library is `ffmpeg`.

Do you have `$PANDIR/timelapse` directory with write permissions? Looks like it doesn't create or check for that but assumes.

wtgee, 2018-01-19:
FWIW, we don't actually use the timelapse for much and currently don't upload them to cloud storage. The jpgs are automatically deleted as part of cleanup so I was creating timelapse as I like to look at them in the morning. They are a quick way to see if any meteors or other odd behavior and also nice to show if tracking problems. But mostly just for fun. We should evaluate their use.

jamessynge, 2018-01-19:
I would like to have these available as a way to do a quick review of the nights observations.

I'll check the file system when I get home.

jamessynge, 2018-01-20:
There was no timelapse directory.

PR #398 (2018-01-20): Create the timelapse directory explicitly.
Use os.path.join instead of string concatenation for building paths.
Fix exception message output.
Add ffmpeg to requirements, needed for timelapse creation.

Fixes #395.
commit ad6115c4 (2018-01-20): Create the timelapse directory explicitly. (#398)

Use os.path.join instead of string concatenation for building paths.
Fix exception message output.
Add ffmpeg to requirements, needed for timelapse creation.

Fixes #395.
closed 2018-01-20 by wtgee