panoptes/POCS #402: Specify constraints in config file

wtgee, 2018-01-21:
The constraints to use are currently hard-coded into the `Observatory` class but need to come from a config file. This would like be part of the move to dependency injection on part of the scheduler.

jermainegug, 2018-01-25:
How are they hard-coded into the `Observaory` class? (Not questioning you, just not sure how to spot this myself)

wtgee, 2018-01-25:
When an `Observatory` object is created it calls out `_create_scheduler`, which has the constraints hard-code [here](https://github.com/panoptes/POCS/blob/develop/pocs/observatory.py#L786). The idea would be to have a list of the constraint names under the `scheduler` key in the config file and then to use the `load_module` (see  [`_create_mount`](https://github.com/panoptes/POCS/blob/develop/pocs/observatory.py#L619) for an example of using this.)

wtgee, 2018-01-25:
Not that you need the names and potentially any options in the config so might be a little tricky.

jermainegug, 2018-01-25:
```
constrains = load_module(
                    'pocs.scheduler.{}'.format(constraint))
```
Do I want to do something like this?

jermainegug, 2018-01-28:
Quite lost with this at the moment, not sure how to do it. Can't understand how `_create_module` works.

wtgee, 2018-01-28:
Ok, I wouldn't worry about this for now. We want to pull all those private create methods out soon anyway and it is probably best done there.

PR #1160 (2022-01-17): Constraints from config file.
* Specify under `scheduler.constraints`
* Some changes to existing constraints for consistency and to make work from config.

Closes #402.
Closes #1057.
Closes #1087.
closed 2022-01-17 by wtgee