panoptes/POCS #416: Horizon info read from config

wtgee, 2018-01-24:


jermainegug, 2018-01-25:
What information are we wanting to put in to the config?

wtgee, 2018-01-26:
It can go into the config as:
```
location:
    name: Macquarie University Observatory
    latitude: -33.7738 # Degrees
    longitude: 151.1126 # Degrees
    elevation: 100.0 # Meters
    utc_offset: 10.00 # Hours
    horizon: 30 # Degrees
    obstructions:
        - [[40, 30], [40, 75]]
        - [[30, 130], [75, 155], [56, 175]]
    twilight_horizon: -18 # Degrees
    timezone: Australia/Sydney
    gmt_offset: 600
```

Then pull the `horizon` and `obstructions` and build the `Horizon` object as something like `Horizon(obstructions, default_horizon)`

wtgee, 2018-01-26:
You can do this while doing #402

jermainegug, 2018-02-11:
Just remembered this is on the panoptes/POCS config, is it okay to temporarily have the obstructions for Huntsman in this config?

jamessynge, 2018-02-11:
I'd prefer not. Can you put it into pocs_local.yaml on Huntsman?

Are you looking for a way to test the handling of that data in a config file?

jermainegug, 2018-02-11:
Yep, that is what I'll do. Thanks.

Just needed to put the information in the config for Huntsman, but I have confused myself as I do not know why I didn't put it into the huntsman config file in the first place.

wtgee, 2018-02-11:
Yes this can directly into https://github.com/AstroHuntsman/huntsman-pocs/blob/develop/conf_files/huntsman.yaml (i.e., *not* the _local version)

wtgee, 2018-04-12:
This has been integrated.

closed 2018-04-12 by wtgee